eez-open/psu-firmware #24: UTFT library usage

prasimix, 2016-10-27:
According to author of UTFT hosting any variant of his library is crime, and since UTFT_Pro is removed from the GitHub we have to remove UTFT_Pro from our distribution and forward users to download library from his site. That will also require the following changes in lcd.cpp:

`#define TFT_320QVT_9341 CTE32_R2`

prasimix, 2017-01-23:
See also #75

mvladic, 2017-03-22:
Following libraries are removed from the project:

- Ethernet2
- UIPEthernet
- UTFT_Pro

We created following forks of third party libraries:

- https://github.com/eez-open/Ethernet2
- https://github.com/eez-open/arduino_uip
- https://github.com/eez-open/scpi-parser

For now, we will keep eez_psu_lib and scpi-parser inside this project.

closed 2017-04-21 by prasimix