eez-open/psu-firmware #35: Channel coupling

prasimix, 2016-11-12:
Implement channel coupling in series and parallel using built in power relays. That require new SCPI commands `INSTrument:COUPle:TRACking` and introducting some sort of command dispatcher that more then one channel can be controlled as single entity. Also we need a new set of main pages where only one set of output programming and measured values will be displayed with new ranges (up to 80 V and up to 10 A). When channels are coupled in series, output voltage on each channel will be set to half of the programmed value. The same is true for setting current when channels are couple in parallel.
Defining coupling mode via local display we can make a wizard with the following steps:
1. Coupling type (series or parallel)
1. Display instruction that connected load has to be disconnected (We can test that by measuring current on both channel. When test passed set VOLT and CURR on both channel to zero, and disable both outputs)
1. Display instruction which binding posts will be used when channels are coupled (CH1-, CH2+)
1. Finished (power relay and SP_LED are activated, down-programmer on CH2 is unconditionally disabled when coupled in parallel and enable both outputs)

As indication when channels are coupled in series and output voltage is set over 40 V the SP_LED could flash. Additional indication could be warning sign on the local display in that mode.

Special case is entering CC mode when channels are couple in series. It's possible that output voltage on one of the channels become negative. To avoid such condition output voltage leveling is required.

prasimix, 2016-11-17:
Tested with 212293d
1. When enter stand-by mode relay is switched off what is fine, but on next power on it's not switched on again
2. When powered on or hard reset relay will be switched on and off

mvladic, 2016-11-18:
Fixed!

prasimix, 2016-11-24:
Add "Not allowed" state for options (as R/O text) that cannot be used while channels are coupled:
- Channel:advanced settings: Low ripple mode (SER only)
- Channel:advanced settings: Remote sensing (SER only)
- Channel:advanced settings: Remote programming
- Channel: information: Calib. params

prasimix, 2016-12-02:
We have to find a way how to display channel specific information while channels are coupled:
1. Current temperature on channels sensors in OTP settings on Protection settings page
1. Total on time, last on time and temperature on Info page

prasimix, 2016-12-04:
OVP trip level cannot be set if channels are coupled in parallel.

prasimix, 2016-12-05:
Waking up from stand-by mode when channels are coupled will not be correct if one of the channel was previously in error mode (e.g. powergood signal failure is detected). Step to reproduce this problem:
1. Couple channels in parallel, set voltage and current, connect load
1. Remove power cable from one Power board, when PSU should go into stand-by
1. Wake it up from stand-by and no current will be sourced to the connected load

closed 2016-12-07 by prasimix