eez-open/psu-firmware #37: Incorrect Power-On under 115-120V AC main supply

algaremi, 2016-11-20:
Hi. I'm having problem with the EEZ using 120V AC main supply, and two SMPS 48V supplies instead of the toroidal transformer. Problem seems to be related with the SMPS to be slower to start under 120Vac, that slow the correct start of both PWR_PCB Pre-Regs and the three digital supplies (+/-5V, 9V). Then the Arduino is doing his startup checks, but doesn't get the replies from the IOEXP and ADC, and put the supply back in Standby, see the USB port trace:

**TRACE [2016-11-18 20:04:48]: PWR_DIRECT -> HIGH
**TRACE [2016-11-18 20:04:48]: Ch1 GPIO 0x0a
**TRACE [2016-11-18 20:04:48]: Ch2 GPIO 0x0a
**TRACE [2016-11-18 20:04:49]: Ch1 GPIO 0x0a
**TRACE [2016-11-18 20:04:49]: Ch2 GPIO 0x0a
**TRACE [2016-11-18 20:04:53]: Ch1 PWRGOOD bit changed to 0
**ERROR [2016-11-18 20:04:53]: -242,"CH1 fault detected"

**TRACE [2016-11-18 20:04:53]: PWR_DIRECT -> LOW
**TRACE [2016-11-18 20:05:04]: PWR_DIRECT -> HIGH
**TRACE [2016-11-18 20:05:04]: Ch1 GPIO 0x0a
**TRACE [2016-11-18 20:05:04]: Ch2 GPIO 0x0a
**TRACE [2016-11-18 20:05:05]: Ch1 GPIO 0x0a
**TRACE [2016-11-18 20:05:05]: Ch2 GPIO 0x0a
**TRACE [2016-11-18 20:05:09]: Ch1 PWRGOOD bit changed to 0
**ERROR [2016-11-18 20:05:09]: -242,"CH1 fault detected"

**TRACE [2016-11-18 20:05:09]: PWR_DIRECT -> LOW

If both PWR_PCB are supplied by an external 48V supply before the EEZ is taking out of standby, then the supply reports no fault, and it is working correctly. Maybe there is the need to add some delay into the code, before doing the selfchecks to let the time for the PWR_PCB supplies to stabilize at startup.

prasimix, 2016-11-21:
Tested with 115 Vac and Mean Well [LRS-150-48](http://www.tme.eu/en/details/lrs-150-48/industrial-power-supplies/mean-well/) and I cannot reproduce this issue. It is tested with 50 Hz and autotransformer, maybe there is some issue with 60 Hz. A further study is required possibly with special firmware version with more debug info.

prasimix, 2016-11-22:
Please try to replace board.cpp with one in the attachment. We added additional 2 seconds delay after power on sequence. If you still have a problem try to increase delay in line 36 (e.g. for 5 seconds set 5000). If this solve your problem try to decrease that delay to the point when issue will appear again and let us know what is that value.

[board.cpp.zip](https://github.com/eez-open/psu-firmware/files/606037/board.cpp.zip)

algaremi, 2016-11-22:
I think I finally found the problem, if not its cause. With the added 
starting delay I had more time to look at the SMPS power supplies 
output. Guess what I found? The problem at startup was always CH1 first, 
then some time CH2. Looking at the scope 48V screen capture we can see 
that the SMPS starts normally, then after the startup delay the CH1 
PWR_PCB starts and the 48V drops drastically, causing probably the 
PWR_PCB three digital supplies to fail as well, that certainly cause the 
IOEXP fault reported by the selfcheck.

The problem is whatever with the SMPS itself, or the CH1 PWR_PCB, that 
seems to be close to short when starting. I'll try the same SMPS module 
with the other CH2 PWR_PCB, then if it is the only the CH1 PWR_PCB, I 
need to get deeper into the circuitry.

Great help guys with this programmable software delay. It made testing 
at startup much easier.

algaremi, 2016-11-22:
Here the 48V trace
[image]

prasimix, 2016-11-23:
Did you get the same glitch when AC/DC module for CH1 is connected directly to the AC mains?

algaremi, 2016-11-23:
Tested with both AC/DC modules connected directly to AC (not going 
through the AUX_PS) and it is working fine. Keep in mind that in this 
case the AC/DC modules are ON before the EEZ PS turns on, like when I'm 
using an external PS. I was not able to reproduce the l=drop on the 
AC/DC module output.

Measure both PWR_PCB DC input current at startup, they are quite low, no 
big current surge, CH1 is 24ma max, 22ma avg, CH2 is 52ma. Nothing to 
severely load the AC/DC modules...

Did also two other tests:
-Supplied both PWR_PCB from the same AC/DC module, each one in turn, no 
change, PS still goes down into standby after startup. No difference 
between either of the AC/DC module used...
-On the AUX_PS, bypassed the 15R of the softstart circuit. Still no 
change, same proble, so it is not causing the AC/DC modules to don't 
start correctly. Remove 15R resistors jumper.

To resume, same problem, EEZ PS working fine if both PWR_PCB are ON 
before the supply controller start, using an external 48Vdc supply, or 
both AC/DC modules, supplied directly from AC line.

Any thought?
SB

prasimix, 2016-11-24:
Please try with attached sketch. It has start up sequence and after that you can see on each outputs something like this:

[image]

[DAC_set_output_voltage_test.zip](https://github.com/eez-open/psu-firmware/files/610890/DAC_set_output_voltage_test.zip)

mvladic, 2016-11-24:
[DAC_set_output_voltage_test.zip](https://github.com/eez-open/psu-firmware/files/610895/DAC_set_output_voltage_test.zip)

mvladic, 2016-11-24:
[DAC_set_output_voltage_test.zip](https://github.com/eez-open/psu-firmware/files/610902/DAC_set_output_voltage_test.zip)

prasimix, 2016-11-24:
Please try if possible also the following scenario: keep AC/DC modules controlled by AUX PS soft-start switch, keep Power boards connected to their AC/DC modules but **disconnect** them from the Arduino shield!  
That will indeed generate self-test errors, but for us is interesting to see if 48 V at the AC/DC modules outputs will have the same huge drop-down or not.

algaremi, 2016-11-24:
I'll try these tests this evening, ThanksSB                                                                                                                                                                                                                                                                                                                                        Sent from my BlackBerry 10 smartphone on the Rogers network.                                                                                                                                                                                                                From: prasimixSent: Thursday, November 24, 2016 5:06 AMTo: eez-open/psu-firmwareReply To: eez-open/psu-firmwareCc: algaremi; AuthorSubject: Re: [eez-open/psu-firmware] Incorrect Power-On under 115-120V AC main supply (#37)Please try if possible also the following scenario: keep AC/DC modules controlled by AUX PS soft-start switch, keep Power boards connected to their AC/DC modules but disconnect them from the Arduino shield!
That will indeed generate self-test errors, but for us is interesting to see if 48 V at the AC/DC modules outputs will have the same huge drop-down or not.

prasimix, 2016-11-30:
Please also let us know if you got the same problem when:
1. both channels are set OFF at the start up and 
1. even if load is not connected.

prasimix, 2016-12-01:
A special [branch](https://github.com/eez-open/psu-firmware/tree/startup-hl) is created with some extra measures that could fix this issue. Delete completely your existing sketch and libraries and download new one.
Please note that this version do not have startup self-test procedure.

algaremi, 2016-12-01:
Ok resuming working on the EEZ supply. First you asked to see if both 
channels are OFF at startup, if the problem was still there. Yes it is, 
supply detects errors and fall back on standby.

algaremi, 2016-12-01:
Second test you asked to check, Both PWR_PCB disconnected from Arduino, 
and both AC/DC supplies still controlled by Arduino Shield, Arduino 
detects all the missing interfaces from both Ch1 & CH2 (Normal...), goes 
in ERROR on the screen and both 48V supplies turns OFF, Arduion doesn't 
go in standby...

algaremi, 2016-12-01:
> dac output test

I tried this new sketch and nothing happens, the 48V supplies don't turn 
ON, so I cannot see the DAC output pattern...
Only the Arduino shield Touchscreen background light turns on, then 
nothing...

SB

algaremi, 2016-12-01:
Clone this new sketch, unzip and tried to upload using Arduino. I got 
the following error at the compilation step:

Arduino: 1.6.12 (Windows 7), Board: "Arduino Due (Programming Port)"

sketch\channel.cpp: In member function 'void 
eez::psu::Channel::tick(long unsigned int)':

channel.cpp:611: error: 'SCPI_ERROR_CH1_DOWN_PROGRAMMER_SWITCHED_OFF' 
was not declared in this scope

psu::generateError(SCPI_ERROR_CH1_DOWN_PROGRAMMER_SWITCHED_OFF + (index 
- 1));

                                         ^

channel.cpp:615: error: 'SCPI_ERROR_CH1_OUTPUT_FAULT_DETECTED' was not 
declared in this scope

psu::generateError(SCPI_ERROR_CH1_OUTPUT_FAULT_DETECTED - (index - 1));

                                         ^

Multiple libraries were found for "scpi-parser.h"
  Used: C:\Users\bergerons\Desktop\Documents\Arduino\libraries\scpi-parser
  Not used: C:\Program Files (x86)\Arduino\libraries\scpi-parser
Multiple libraries were found for "eez_psu_rev.h"
  Used: C:\Users\bergerons\Desktop\Documents\Arduino\libraries\eez_psu_lib
  Not used: C:\Program Files (x86)\Arduino\libraries\eez_psu_lib
  Not used: C:\Program Files (x86)\Arduino\libraries\eez_psu_lib
  Not used: C:\Program Files (x86)\Arduino\libraries\eez_psu_lib
  Not used: C:\Program Files (x86)\Arduino\libraries\eez_psu_lib
Multiple libraries were found for "Ethernet2.h"
  Used: C:\Users\bergerons\Desktop\Documents\Arduino\libraries\Ethernet2
  Not used: C:\Program Files (x86)\Arduino\libraries\Ethernet2
Multiple libraries were found for "UTFT.h"
  Used: C:\Users\bergerons\Desktop\Documents\Arduino\libraries\UTFT_Pro
  Not used: C:\Program Files (x86)\Arduino\libraries\UTFT_Pro
exit status 1
*'SCPI_ERROR_CH1_DOWN_PROGRAMMER_SWITCHED_OFF*' was not declared in this 
scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

algaremi, 2016-12-01:
Well something seems to be broken with my Arduino, tried to re-installed 
the original sketch, but got an other libraries error. Today I autorized 
libraries update... I tried to de-install the Arduino program, and 
re-install from scratch but it doens't seem to work, all my previous 
settings are still there... Any idea how I can solved this problem?
Thanks
SB

Arduino: 1.6.13 (Windows 7), Board: "Arduino Due (Programming Port)"

sketch\ethernet.cpp: In function 'bool eez::psu::ethernet::init()':

ethernet.cpp:140: error: 'class EthernetClass' has no member named 
'setDhcpTimeout'

      Ethernet.setDhcpTimeout(ETHERNET_DHCP_TIMEOUT * 1000UL);

               ^

exit status 1
'class EthernetClass' has no member named 'setDhcpTimeout'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

algaremi, 2016-12-01:
Ok fixed my Arduion, re-installed the version 1.6.9 (not the latest 
one), copied missing EEZ librairies to the Arduino Library folder, now 
the original master version M2 compile and upload without error. I'll 
try the new program path for Heavy load.

prasimix, 2016-12-01:
`'SCPI_ERROR_CH1_DOWN_PROGRAMMER_SWITCHED_OFF' was not declared in this scope` means that you didn't copy libraries required for firmware sketch to compile. You need to do that manually as first time.

algaremi, 2016-12-01:
Ok, now a strange problem, the new master-hl compile fine, but it is not 
uploading completely. It stops during the upload process and stay there 
hanging, doing nothing. The Arduino USB port led stops flashing...

Just to be sure I used the previous Master sketch and the one compile 
and upload just fine...

I retried three times, tried different USB ports on my laptop, no 
success. For unknown reason the upload is not working with this 
version... The Arduino didn't report error during the upload process.

SB

algaremi, 2016-12-01:
Screen capture of the Arduino, frozen in the Uplaod process:
[image]

prasimix, 2016-12-01:
Could you please detach Arduino Due from the shield and try upload it. Also be sure that programming port is used. In Arduino IDE select **Tools ... Board ... Board manager** and please let us know what _Arduino SAM boards_ version you have installed.

algaremi, 2016-12-02:
Tried with the Arduino by itself and same problem...

prasimix, 2016-12-02:
Ok, I tried it here with both IDE 1.6.9 and 1.6.13 just to be sure that the latest version has some files that remains even if you goes back to 1.6.9. Everything works fine but I'm on Linux (if that makes any difference).
Hopefully you restart Windows (after each upload failure) to be sure that something is not stuck in USB or some other lower level.

prasimix, 2016-12-02:
Without further struggling with newer version of sketch, please upload M2, remove jumper JP8 on Arduino Shield and let us know what's going on.

algaremi, 2016-12-02:
Hi. I rebooted my laptop, no change, and I was wrong, the upload stops 
always at the same memory location, check the recording of three try I 
did, it stopped always at the same point, I'll go back to M2 then

readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x20001024

writeWord(addr=0x2000102c,value=0xbf100)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x20001024,value=0x20001001)

readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x20001024,value=0x20001001)

prasimix, 2016-12-02:
Ok, please try the new build a47e2ff on the master branch.

algaremi, 2016-12-02:
It is working! Like miracle everything is nowworking just fine, standby 
mode leaves both AC/DC converters ON, front power switch shut down both 
AC/DC converters, both PWR_PCB are working just fine, supplying 1A in CC 
mode for CH2 and 22V, 0.35Amp on CH1 at the moment, 5mm gel coupling 
heatsink pads finally received yesterday and installed, the PWR_PCB are 
just barely warm, perfect!

What is JP8 doing, or not doing?

Seems to have solved my problem entirely!

Thanks
SB

prasimix, 2016-12-02:
Ok, if you remove it now please put it back and try the latest firmware! This issue is not related to 120VAC but it is just sooner manifested with lower mains voltage. Ironically this is a guard against MCU blocking, but since WATCHDOG signal is not properly generated on the start it pull down AC power. I'll change the name of this issue.

algaremi, 2016-12-02:
Do I need to re-install JP8 with this built?

prasimix, 2016-12-02:
Yes, idea is to make it work with JP8 installed. That was firmware, not hardware bug.

algaremi, 2016-12-02:
Ok I'll put it back, so it was finally really a software bug, glad to 
see that I was dreaming it!

BTW how is the fan controlled? I tried the low ripple feature on channel 
1, and the PWR module temp went up and exceeded 50C, and the OTC 
protection kick-in, but the fan never started, which makes me a little 
worried...

SB

algaremi, 2016-12-02:
BTW how do I download the new built, seems to have multiple files 
changed. Do I simply re-download the complete M2 and re-install it?

Thanks
SB

prasimix, 2016-12-02:
The simplest way to follow new build is to install `git` and synchronize with master repository by entering:
`git clone https://github.com/eez-open/psu-firmware`
And for each new build you need to enter:
`git pull`

If that is too complicated simply download everything from master branch and overwrite previous local copy.

algaremi, 2016-12-02:
Installed and used Git and worked like a charm, Downloaded latest built, 
but again same bug, jammed during the upload to Arduino.
I cannot install any othe r built than the original M2. Something 
changed since then, very strange. I'm using Arduino 1.6.9. Should I 
upgrade to the latest Arduino?

SB

prasimix, 2016-12-02:
Yes, that's strange. Do you have problem with compiling or uploading?

algaremi, 2016-12-02:
Compiling/Verifying is working fine. Problem only with uploading, but 
original M2 is working fine...

prasimix, 2016-12-02:
Please note that M3 is much larger then M2 and require more time for upload. I don't know if `Show verbose output during upload` helps or make thigs even slower.

algaremi, 2016-12-02:
Longer is one thing, but the upload just freeze, see latest screen 
capture, you see no progress bar or nothing:
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWo

I'll try with verbose Off

algaremi, 2016-12-02:
Tried verbose off, and the upload jam during the verify phase, see screen capture...

[image]

prasimix, 2016-12-02:
Congratulations! It seems that you just find bug in Arduino IDE for Due.
Please switch off `Verify code after upload`! As you can see upload is successfully finished.

algaremi, 2016-12-02:
Yes Sir it worked. I was able to upload the code with |Verify code after 
upload|! set to Off.

Now I have M3 code, but still a problem, JP8 re-installed, supply turns 
on, but report both channels in ERROR and turn off both AC/DC modules...

Back to square one it seems...

SB

algaremi, 2016-12-02:
Errors reported at startup are both CH1 & CH2 IOEXP & ADC failed, but at 
least the supply doesn't go in standby immediately...

SB

prasimix, 2016-12-02:
Ok, please check once again that you are downloading the latest build. Then use one of serial monitor which allows you to copy&paste serial output (check [here](https://www.baldengineer.com/alternatives-to-arduinos-serial-monitor.html)). After downloading latest build add this code into conf_user.h:
```
/// Interval (in milliseconds) at which watchdog impulse will be sent
#undef WATCHDOG_INTERVAL
#define WATCHDOG_INTERVAL 500
```
Send us what you receive on start up.

algaremi, 2016-12-02:
I have built: psu-firmware-a47e2ff854e657dd876232ff481d7a4c55480d54

Here the code at startup:
EEZ PSU serial com ready
**TRACE: Touch screen calibration points: 3593, 468, 429, 3575, 449, 511
**TRACE: Touch modification ENABLED
**TRACE [2016-12-02 14:48:51]: PWR_DIRECT -> HIGH
**TRACE [2016-12-02 14:48:52]: BP 0x0000
**TRACE [2016-12-02 14:48:52]: Ch1 GPIO 0x0a
**TRACE [2016-12-02 14:48:52]: Ch2 GPIO 0x0a
**TRACE [2016-12-02 14:48:52]: Ch1 IO expander reg check failure: reg=0, 
expected=101, got=255
**ERROR [2016-12-02 14:48:52]: 210,"CH1 IOEXP test failed"

**TRACE [2016-12-02 14:48:52]: Ch1 ADC test failed reg1: expected=160, 
got=255
**TRACE [2016-12-02 14:48:52]: Ch1 ADC test failed reg2: expected=96, 
got=255
**TRACE [2016-12-02 14:48:52]: Ch1 ADC test failed reg3: expected=0, got=255
**ERROR [2016-12-02 14:48:52]: 220,"CH1 ADC test failed"

**TRACE [2016-12-02 14:48:52]: Ch1 DAC test skipped because of IO expander
**TRACE [2016-12-02 14:48:52]: Ch2 IO expander reg check failure: reg=0, 
expected=101, got=255
**ERROR [2016-12-02 14:48:52]: 211,"CH2 IOEXP test failed"

**TRACE [2016-12-02 14:48:52]: Ch2 ADC test failed reg1: expected=160, 
got=255
**TRACE [2016-12-02 14:48:52]: Ch2 ADC test failed reg2: expected=96, 
got=255
**TRACE [2016-12-02 14:48:52]: Ch2 ADC test failed reg3: expected=0, got=255
**ERROR [2016-12-02 14:48:53]: 221,"CH2 ADC test failed"

**TRACE [2016-12-02 14:48:53]: Ch2 DAC test skipped because of IO expander
**TRACE [2016-12-02 14:48:53]: Fan RPM: 4317

SB

prasimix, 2016-12-03:
Bug with Arduino IDE is reported [here](https://github.com/arduino/Arduino/issues/5672).

This is now starts to be weird. Please decrease `WATCHDOG` interval to 500 ms as suggested in my previous comment and send us if possible on the single screen signal traces of:
1. +48 VDC input on any of Power boards
1. Arduino Due pin 5 (it's 3rd pin from the left on the second header on the upper row of headers if you are looking from the front panel side)

algaremi, 2016-12-03:
>  2. Arduino Due pin 5 (it's 3rd pin from the left on the second header

Ok, probably didn't copy the watchdog 500ms instructions in conf_user.h 
the last firmware pull I did. Forgot, it was late yesterday, long day...
Pull latest firmware "M3 (a47e2ff)", added watchdog 500ms, compile and 
upload.

Ok this time still has the PWR_PCB faults, but the 48V doesn't even turn 
on at startup, no wonder the PWR_PCB are not working, there is no supply...

Trace at startup, going out of standby:
**TRACE [2016-12-03 13:47:32]: PWR_DIRECT -> HIGH
**TRACE [2016-12-03 13:47:32]: BP 0x0000
**TRACE [2016-12-03 13:47:32]: Ch1 GPIO 0x0a
**TRACE [2016-12-03 13:47:32]: Ch2 GPIO 0x0a
**TRACE [2016-12-03 13:47:32]: Ch1 IO expander reg check failure: reg=0, 
expected=101, got=255
**ERROR [2016-12-03 13:47:32]: 210,"CH1 IOEXP test failed"

**TRACE [2016-12-03 13:47:32]: Ch1 ADC test failed reg1: expected=160, 
got=255
**TRACE [2016-12-03 13:47:32]: Ch1 ADC test failed reg2: expected=96, 
got=255
**TRACE [2016-12-03 13:47:32]: Ch1 ADC test failed reg3: expected=0, got=255
**ERROR [2016-12-03 13:47:32]: 220,"CH1 ADC test failed"

**TRACE [2016-12-03 13:47:32]: Ch1 DAC test skipped because of IO expander
**TRACE [2016-12-03 13:47:32]: Ch2 IO expander reg check failure: reg=0, 
expected=101, got=255
**ERROR [2016-12-03 13:47:32]: 211,"CH2 IOEXP test failed"

**TRACE [2016-12-03 13:47:32]: Ch2 ADC test failed reg1: expected=160, 
got=255
**TRACE [2016-12-03 13:47:32]: Ch2 ADC test failed reg2: expected=96, 
got=255
**TRACE [2016-12-03 13:47:33]: Ch2 ADC test failed reg3: expected=0, got=255
**ERROR [2016-12-03 13:47:33]: 221,"CH2 ADC test failed"

**TRACE [2016-12-03 13:47:33]: Ch2 DAC test skipped because of IO expander
**TRACE [2016-12-03 13:47:33]: Fan RPM: 4317

SB

algaremi, 2016-12-03:
Upgraded Arduino Due board driver to 1.6.9, recompile and tried 
different Watchdog delays from 100 to 900ms, and same problem, the AC/DC 
modules don't start with this version...

SB

prasimix, 2016-12-03:
Is it possible to send us screenshot of pin 5 on Arduino board as I suggested in my latest comment? You can leave `WATCHDOG_INTERVAL` to default (1000).

algaremi, 2016-12-03:
Ok, will do tomorrow, thanks

algaremi, 2016-12-05:
Hi. As requested here is pin 5 on the Arduino scope screen capture, there a short pulse (PW: 3 uS). Is it telling you anything?
[image]

prasimix, 2016-12-05:
Yes, that's WATCHDOG signal. Can you make another screenshot of the start-up with much longer timebase (e.g. 1 sec)? If it for some reason is missing for more then about 1.6 seconds AC power will be turned off (with JP8 installed).

algaremi, 2016-12-05:
Will do in a few hours. thanks

SB

algaremi, 2016-12-06:
Here Arduino pin5 scope capture from power-on
[image]

algaremi, 2016-12-06:
Updated firmware to latest version with git, redo screen capture, pulse had changed
[image]

prasimix, 2016-12-06:
Ok, it seems that you don't have continuous generation of watchdog pulses and therefore nothing is happen on the start. Could you please add into your conf_user.h file the following code (or uncomment it if it's already there as part of some of recent builds):
```
#undef OPTION_WATCHDOG
#define OPTION_WATCHDOG 0
```
That should disable generation of watchdog signal and TPS3705-33 should not affect AC soft-start control.

algaremi, 2016-12-06:
But the real question is why no continuos pulses. Are the pulses generated by IC32 or the Arduino code?
SB

algaremi, 2016-12-07:
Uncommented both lines, recompiled and upload. Still not working, both AC/DC modules stays off...

prasimix, 2016-12-07:
Please find below how it looks like on the my end with the latest build when power on is initiated:

[image]

If you still have a problem, send us your `conf_user.h`, but if there is nothing questionable we don't know how to proceed with this issue then wait for other people who built their unit to see if they'll report the same issue.  
In the meantime simply remove JP8.

[3 more replies]

closed 2016-12-07 by algaremi