eez-open/psu-firmware #44: Suggested user profile description when channels are coupled

prasimix, 2016-12-09:
When params are saved into unused user profile suggested description is not correct.

algaremi, 2016-12-17:
I cannot save any user profile from 1 to 9. Power-On profile 0 is working fine. Also I had no problem doing the calibration routine.

mvladic, 2016-12-19:
I added some debug trace logging that should help us investigate the problem. Extract `gui_page_user_profiles.cpp` from the [gui_page_user_profiles.zip](https://github.com/eez-open/psu-firmware/files/661190/gui_page_user_profiles.zip)
 and save it inside `eez_psu_sketch` folder. Try again to save into unused user profile and send me the log from the serial monitor. Thanks!

algaremi, 2016-12-19:
Hi. Here the trace just after coming out of standby. Thanks
SB

**TRACE [2016-12-19 10:26:10]: PWR_DIRECT -> HIGH
**TRACE [2016-12-19 10:26:10]: Ch1 GPIO 0x0a
**TRACE [2016-12-19 10:26:10]: Ch2 GPIO 0x0a
**TRACE [2016-12-19 10:26:11]: Fan RPM: 4174
**TRACE [2016-12-19 10:26:19]: Save profile T1
**TRACE [2016-12-19 10:26:19]: Save profile T2
**TRACE [2016-12-19 10:26:19]: Save profile T4
**TRACE [2016-12-

mvladic, 2016-12-19:
Is every time the same result?

mvladic, 2016-12-19:
And please try with the latest version on github.

algaremi, 2016-12-19:
Yes

algaremi, 2016-12-19:
Just installed the latest built and still not working, latest trace:

**TRACE [2016-12-19 11:30:59]: PWR_DIRECT -> HIGH
**TRACE [2016-12-19 11:30:59]: Ch1 GPIO 0x0a
**TRACE [2016-12-19 11:30:59]: Ch2 GPIO 0x0a
**TRACE [2016-12-19 11:31:00]: Fan RPM: 4260
**TRACE [2016-12-19 11:31:08]: Save profile T1
**TRACE [2016-12-19 11:31:08]: Save profile T2
**TRACE [2016-12-19 11:31:08]: Save profile T4
**TRACE [2016-12-19 11:31:08]: Save profile T5
**TRACE [2016-12-19 11:31:08]: Save profile T6
**TRACE [2016-12-19 11:31:08]: Save profile T7

mvladic, 2016-12-19:
Thanks! Please try once more with the latest version from the github - I added a few more trace lines.

algaremi, 2016-12-19:
Here the trace of the latest built:

**TRACE [2016-12-19 14:50:32]: PWR_DIRECT -> HIGH
**TRACE [2016-12-19 14:50:32]: Ch1 GPIO 0x0a
**TRACE [2016-12-19 14:50:32]: Ch2 GPIO 0x0a
**TRACE [2016-12-19 14:50:33]: Fan RPM: 4230
**TRACE [2016-12-19 14:50:38]: Save profile T7.2.1
**TRACE [2016-12-19 14:50:38]: Save profile T7.2.2
**TRACE [2016-12-19 14:50:38]: Save profile T7.2.8
**TRACE [2016-12-19 14:50:38]: Save profile T7.2.9
**TRACE [2016-12-19 14:50:38]: Save profile T7.2.10
**TRACE [2016-12-19 14:50:38]: Save profile T7.2.11
**TRACE [2016-12-19 14:50:41]: Save profile T1
**TRACE [2016-12-19 14:50:41]: Save profile T2
**TRACE [2016-12-19 14:50:41]: Save profile T4
**TRACE [2016-12-19 14:50:41]: Save profile T5
**TRACE [2016-12-19 14:50:41]: Save profile T6
**TRACE [2016-12-19 14:50:41]: Save profile T7
**TRACE [2016-12-19 14:50:41]: Save profile T7.1
**TRACE [2016-12-19 14:50:41]: Save profile T7.1.1
**TRACE [2016-12-19 14:50:41]: Save profile T7.1.2
**TRACE [2016-12-19 14:50:41]: Save profile T7.1.3
**TRACE [2016-12-19 14:50:41]: Save profile T7.1.5
**TRACE [2016-12-19 14:50:42]: Save profile T7.2
**TRACE [2016-12-19 14:50:42]: Save profile T7.2.1
**TRACE [2016-12-19 14:50:42]: Save profile T7.2.2
**TRACE [2016-12-19 14:50:42]: Save profile T7.2.8
**TRACE [2016-12-19 14:50:42]: Save profile T7.2.9
**TRACE [2016-12-19 14:50:42]: Save profile T7.2.10

algaremi, 2016-12-19:
**TRACE [2016-12-19 14:50:32]: PWR_DIRECT -> HIGH
**TRACE [2016-12-19 14:50:32]: Ch1 GPIO 0x0a
**TRACE [2016-12-19 14:50:32]: Ch2 GPIO 0x0a
**TRACE [2016-12-19 14:50:33]: Fan RPM: 4230
**TRACE [2016-12-19 14:50:38]: Save profile T7.2.1
**TRACE [2016-12-19 14:50:38]: Save profile T7.2.2
**TRACE [2016-12-19 14:50:38]: Save profile T7.2.8
**TRACE [2016-12-19 14:50:38]: Save profile T7.2.9
**TRACE [2016-12-19 14:50:38]: Save profile T7.2.10
**TRACE [2016-12-19 14:50:38]: Save profile T7.2.11
**TRACE [2016-12-19 14:50:41]: Save profile T1
**TRACE [2016-12-19 14:50:41]: Save profile T2
**TRACE [2016-12-19 14:50:41]: Save profile T4
**TRACE [2016-12-19 14:50:41]: Save profile T5
**TRACE [2016-12-19 14:50:41]: Save profile T6
**TRACE [2016-12-19 14:50:41]: Save profile T7
**TRACE [2016-12-19 14:50:41]: Save profile T7.1
**TRACE [2016-12-19 14:50:41]: Save profile T7.1.1
**TRACE [2016-12-19 14:50:41]: Save profile T7.1.2
**TRACE [2016-12-19 14:50:41]: Save profile T7.1.3
**TRACE [2016-12-19 14:50:41]: Save profile T7.1.5
**TRACE [2016-12-19 14:50:42]: Save profile T7.2
**TRACE [2016-12-19 14:50:42]: Save profile T7.2.1
**TRACE [2016-12-19 14:50:42]: Save profile T7.2.2
**TRACE [2016-12-19 14:50:42]: Save profile T7.2.8
**TRACE [2016-12-19 14:50:42]: Save profile T7.2.9
**TRACE [2016-12-19 14:50:42]: Save profile T7.2.10

mvladic, 2016-12-19:
We are getting closer ;) Try again with the latest version. Thanks!

algaremi, 2016-12-20:
More progress in the trace but still not working. But I agree there is progress ;)

**TRACE [2016-12-19 19:01:13]: Ch1 GPIO 0x8a
**TRACE [2016-12-19 19:01:13]: Ch1 GPIO 0x88
**TRACE [2016-12-19 19:01:13]: Ch2 GPIO 0x8a
**TRACE [2016-12-19 19:01:13]: Ch2 GPIO 0x88
**TRACE [2016-12-19 19:01:16]: Save profile T7.2.10.1
**TRACE [2016-12-19 19:01:16]: Save profile T7.2.10.2
**TRACE [2016-12-19 19:01:16]: Save profile T7.2.10.3
**TRACE [2016-12-19 19:01:16]: Save profile T7.2.10.4
**TRACE [2016-12-19 19:01:16]: Save profile T7.2.10.5
**TRACE [2016-12-19 19:01:16]: Save profile T7.2.10.6
**TRACE [2016-12-19 19:01:16]: Save profile T7.2.10.7
**TRACE [2016-12-19 19:01:16]: Save profile T7.2.10.8
**TRACE [2016-12-19 19:01:17]: Channel voltage balancing: CH1_Umon=6.328516, CH2_Umon=-0.672488
**TRACE [2016-12-19 19:01:22]: Save profile T7.2.1
**TRACE [2016-12-19 19:01:22]: Save profile T7.2.2
**TRACE [2016-12-19 19:01:22]: Save profile T7.2.8
**TRACE [2016-12-19 19:01:22]: Save profile T7.2.9
**TRACE [2016-12-19 19:01:22]: Save profile T7.2.10
**TRACE [2016-12-19 19:01:22]: Save profile T7.2.10.1
**TRACE [2016-12-19 19:01:22]: Save profile T7.2.10.2
**TRACE [2016-12-19 19:01:22]: Save profile T7.2.10.3
**TRACE [2016-12-19 19:01:22]: Save profile T7.2.10.4
**TRACE [2016-12-19 19:01:22]: Save profile T7.2.10.5
**TRACE [2016-12-19 19:01:23]: Save profile T7.2.10.6
**TRACE [2016-12-19 19:01:23]: Save profile T7.2.10.7
**TRACE [2016-12-19 19:01:23]: Save profile T7.2.10.8
**TRACE [2016-12-19 19:01:23]: Save profile T7.2.11
**TRACE [2016-12-19 19:01:25]: Save profile T1
**TRACE [2016-12-19 19:01:25]: Save profile T2
**TRACE [2016-12-19 19:01:25]: Save profile T4
**TRACE [2016-12-19 19:01:25]: Save profile T5
**TRACE [2016-12-19 19:01:25]: Save profile T6
**TRACE [2016-12-19 19:01:25]: Save profile T7
**TRACE [2016-12-19 19:01:25]: Save profile T7.1
**TRACE [2016-12-19 19:01:25]: Save profile T7.1.1
**TRACE [2016-12-19 19:01:25]: Save profile T7.1.2
**TRACE [2016-12-19 19:01:25]: Save profile T7.1.3
**TRACE [2016-12-19 19:01:26]: Save profile T7.1.5
**TRACE [2016-12-19 19:01:26]: Save profile T7.2
**TRACE [2016-12-19 19:01:26]: Save profile T7.2.1
**TRACE [2016-12-19 19:01:26]: Save profile T7.2.2
**TRACE [2016-12-19 19:01:26]: Save profile T7.2.8
**TRACE [2016-12-19 19:01:26]: Save profile T7.2.9
**TRACE [2016-12-19 19:01:26]: Save profile T7.2.10
**TRACE [2016-12-19 19:01:26]: Save profile T7.2.10.1
**TRACE [2016-12-19 19:01:26]: Save profile T7.2.10.2
**TRACE [2016-12-19 19:01:26]: Save profile T7.2.10.3
**TRACE [2016-12-19 19:01:26]: Save profile T7.2.10.4
**TRACE [2016-12-19 19:01:26]: Save profile T7.2.10.5
**TRACE [2016-12-19 19:01:26]: Save profile T7.2.10.6
**TRACE [2016-12-19 19:01:26]: Save profile T7.2.10.7

mvladic, 2016-12-20:
Please try once more with these:

[eez_psu_sketch.zip](https://github.com/eez-open/psu-firmware/files/663480/eez_psu_sketch.zip)

algaremi, 2016-12-21:
Problem with Couped Channel, Serial mode, here the trace:
**TRACE [2016-12-21 10:24:22]: Ch1 GPIO 0x8a
**TRACE [2016-12-21 10:24:22]: Ch1 GPIO 0x88
**TRACE [2016-12-21 10:24:22]: Ch2 GPIO 0x8a
**TRACE [2016-12-21 10:24:22]: Ch2 GPIO 0x88
**TRACE [2016-12-21 10:24:24]: Channel voltage balancing: CH1_Umon=6.349267, CH2_Umon=-0.667605
**TRACE [2016-12-21 10:24:32]: Last watchdog impulse was before 66352 uS
**TRACE [2016-12-21 10:24:32]: Ch1 PWRGOOD bit changed to 0
**ERROR [2016-12-21 10:24:32]: -242,"CH1 fault detected"

**TRACE [2016-12-21 10:24:32]: Ch1 GPIO 0x08
**TRACE [2016-12-21 10:24:32]: Restore voltage to value before balancing: 20.000000
**TRACE [2016-12-21 10:24:32]: Ch2 GPIO 0x08
**TRACE [2016-12-21 10:24:33]: Last watchdog impulse was before 681307 uS
**TRACE [2016-12-21 10:24:33]: PWR_DIRECT -> LOW

algaremi, 2016-12-21:
Loaded latest built, put back gui_page_user_profiles.cpp file, tried to save. There is less trace steps:
**TRACE [2016-12-21 10:33:17]: Ch1 GPIO 0x8a
**TRACE [2016-12-21 10:33:18]: Ch1 GPIO 0x88
**TRACE [2016-12-21 10:33:18]: Ch2 GPIO 0x8a
**TRACE [2016-12-21 10:33:18]: Ch2 GPIO 0x88
**TRACE [2016-12-21 10:33:24]: Save profile T1
**TRACE [2016-12-21 10:33:24]: Save profile T2
**TRACE [2016-12-21 10:33:24]: Save profile T4
**TRACE [2016-12-21 10:33:24]: Save profile T5
**TRACE [2016-12-21 10:33:24]: Save profile T6
**TRACE [2016-12-21 10:33:24]: Save profile T7

mvladic, 2016-12-21:
Please try with the files from the zip file I attached in my last comment from two days ago.

algaremi, 2016-12-21:
Sorry I lost track of time in the last two days, I was sick and stuck to 
bed, Felling still weak, but I'm trying the files at this moment. Will 
report in a few minutes. Thanks
SB

algaremi, 2016-12-21:
Sorry, retried with the proper traces files, here the trace:
**TRACE [2016-12-21 11:05:48]: Ch1 GPIO 0x8a
**TRACE [2016-12-21 11:05:48]: Ch1 GPIO 0x88
**TRACE [2016-12-21 11:05:49]: Ch1 GPIO 0x08
**TRACE [2016-12-21 11:05:50]: Ch1 GPIO 0x0a
**TRACE [2016-12-21 11:05:54]: Save profile T1
**TRACE [2016-12-21 11:05:54]: T1
**TRACE [2016-12-21 11:05:54]: T2
**TRACE [2016-12-21 11:05:54]: T2.1
**TRACE [2016-12-21 11:05:54]: T2.2
**TRACE [2016-12-21 11:05:54]: T2.3
**TRACE [2016-12-21 11:05:54]: T2.3
**TRACE [2016-12-21 11:05:54]: T2.4
**TRACE [2016-12-21 11:05:54]: T2.5
**TRACE [2016-12-21 11:05:54]: T2.6
**TRACE [2016-12-21 11:05:54]: T2.7

mvladic, 2016-12-21:
Once more, please.

[eez_psu_sketch.zip](https://github.com/eez-open/psu-firmware/files/666852/eez_psu_sketch.zip)

algaremi, 2016-12-21:
Here with the latest files:

**TRACE [2016-12-21 11:39:13]: Ch1 GPIO 0x08
**TRACE [2016-12-21 11:39:13]: Ch1 GPIO 0x0a
**TRACE [2016-12-21 11:39:13]: Ch2 GPIO 0x08
**TRACE [2016-12-21 11:39:13]: Ch2 GPIO 0x0a
**TRACE [2016-12-21 11:39:19]: T2.7
**TRACE [2016-12-21 11:39:20]: T2.7.1
**TRACE [2016-12-21 11:39:20]: T2.7.2

mvladic, 2016-12-21:
Now?

[eez_psu_sketch.zip](https://github.com/eez-open/psu-firmware/files/666903/eez_psu_sketch.zip)

algaremi, 2016-12-21:
Trace:
**TRACE [2016-12-21 12:17:10]: Ch1 GPIO 0x8a
**TRACE [2016-12-21 12:17:10]: Ch1 GPIO 0x88
**TRACE [2016-12-21 12:17:10]: Ch2 GPIO 0x8a
**TRACE [2016-12-21 12:17:10]: Ch2 GPIO 0x88
**TRACE [2016-12-21 12:17:15]: T2.7
**TRACE [2016-12-21 12:17:15]: T2.7.1
**TRACE [2016-12-21 12:17:15]: T2.7.2 1
**TRACE [2016-12-21 12:17:15]: T2.7.3
**TRACE [2016-12-21 12:17:15]: T2.7.4
**TRACE [2016-12-21 12:17:15]: T2.7.5
**TRACE [2016-12-21 12:17:15]: T2.7.7
**TRACE [2016-12-21 12:17:15]: T2.7.8
**TRACE [2016-12-21 12:17:15]: T2.7.11
**TRACE [2016-12-21 12:17:15]: T2.7.12
**TRACE [2016-12-21 12:17:15]: T2.7.12
**TRACE [2016-12-21 12:17:15]: T2.8
**TRACE [2016-12-21 12:17:15]: T2.8.1
**TRACE [2016-12-21 12:17:15]: T2.8.6
**TRACE [2016-12-21 12:17:16]: T2.9

mvladic, 2016-12-21:
Still doesn't work? Is there any change compared how it was at the beginning when you first reported this problem?

algaremi, 2016-12-21:
No same problem, I cannot save any user profile.

SB

mvladic, 2016-12-21:
When you press the save button do you see the onscreen keyboard or it freezes immediately after you press the save button?

Please try again with these:

[eez_psu_sketch.zip](https://github.com/eez-open/psu-firmware/files/667216/eez_psu_sketch.zip)

algaremi, 2016-12-22:
Hi. No I don't see the onscreen keyboard, it freezes  immediately after you press the save button. I'm trying your file now. Thanks

algaremi, 2016-12-22:
The trace is somewhat different now, see below. Again I haven't seen the onscreen keyboard. Thanks

**TRACE [2016-12-22 12:19:32]: Ch1 GPIO 0x8a
**TRACE [2016-12-22 12:19:32]: Ch1 GPIO 0x88
**TRACE [2016-12-22 12:19:44]: T01
**TRACE [2016-12-22 12:19:44]: T0.1: 0
**TRACE [2016-12-22 12:19:45]: T0.2: 0
**TRACE [2016-12-22 12:19:45]: T0.3
**TRACE [2016-12-22 12:19:45]: T0.4: 0
**TRACE [2016-12-22 12:19:45]: T2.7.1
**TRACE [2016-12-22 12:19:45]: T2.7.2: -1
**TRACE [2016-12-22 12:19:45]: T02

mvladic, 2016-12-22:
Did you changed anything in conf_user.h?

mvladic, 2016-12-22:
Please try this: 
[gui_keypad.zip](https://github.com/eez-open/psu-firmware/files/669415/gui_keypad.zip)

algaremi, 2016-12-22:
I used to changed Watchdog parameters, but I think I have the standard version now, Here what I have: /** @file conf_user.h
@brief Configuration overrided by the user.
Use this header file to override anything from conf.h or conf_advanced.h.
*/

#pragma once

#define ETHERNET_MAC_ADDRESS { 0x74, 0x69, 0x69, 0x2D, 0x30, 0x00 }

// Example: redefine channels
/*
#undef CHANNELS
#define CHANNELS \
    CHANNEL(1, CH_BOARD_REVISION_R5B9_PARAMS, CH_PINS_1, CH_PARAMS_50V_3A), \
    CHANNEL(2, CH_BOARD_REVISION_R5B9_PARAMS, CH_PINS_2, CH_PARAMS_50V_3A) \
*/
/// Firmware version.
#undef FIRMWARE
#define FIRMWARE     "M3 (48554b5)"

/*
#undef ADC_USE_INTERRUPTS
#define ADC_USE_INTERRUPTS 0

/// 0: 20 SPS, 1: 45 SPS, 2: 90 SPS, 3: 175 SPS, 4: 330 SPS, 5: 600 SPS, 6: 1000 SPS
#undef ADC_SPS
#define ADC_SPS 5

#if !ADC_USE_INTERRUPTS
#define ADC_READ_TIME_US 2000L
#endif
*/

//#undef OPTION_WATCHDOG
//#define OPTION_WATCHDOG 0

/// Max. allowed output current (in ampers) if fan or temp. sensor is invalid.
#undef ERR_MAX_CURRENT 
#define ERR_MAX_CURRENT 2

/// Interval (in milliseconds) at which watchdog impulse will be sent
#undef WATCHDOG_INTERVAL
#define WATCHDOG_INTERVAL 250

algaremi, 2016-12-22:
Still not working, no keyboard, but new trace:
**TRACE [2016-12-22 12:55:20]: Ch1 GPIO 0x8a
**TRACE [2016-12-22 12:55:20]: Ch1 GPIO 0x88
**TRACE [2016-12-22 12:55:25]: T01
**TRACE [2016-12-22 12:55:25]: T0.1: 0
**TRACE [2016-12-22 12:55:25]: T0.2: 0
**TRACE [2016-12-22 12:55:25]: T0.3
**TRACE [2016-12-22 12:55:25]: T0.4: 0
**TRACE [2016-12-22 12:55:25]: T0.5: 0
**TRACE [2016-12-22 12:55:25]: T0.6: 0
**TRACE [2016-12-22 12:55:25]: T0.7: 222
**TRACE [2016-12-22 12:55:25]: ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿá
**TRACE [2016-12-22 12:55:26]: T0.8: -1
**TRACE [2016-12-22 12:55:26]: T2.7.1
**TRACE [2016-12-22 12:55:26]: T2.7.2: -1
**TRACE [2016-12-22 12:55:26]: T02

mvladic, 2016-12-22:
Once more:
[profile.zip](https://github.com/eez-open/psu-firmware/files/669466/profile.zip)

algaremi, 2016-12-22:
It Worked! Everything seems to work now, just like magic ;) Tried to save 3 user profile so far with independant supplies or parallel couplig, and recalled either of the saved profiles. Supply reconfigured according to the save profile without problem. Also tried to delete a profile and it worked as well. The bug seems to have been completly solved. Thanks guys.

Here the trace of the save ones, it case it may be of some help. Do you leave the trace there, or it will be corrected in the next built?

**TRACE [2016-12-22 13:15:49]: Ch1 GPIO 0x8a
**TRACE [2016-12-22 13:15:49]: Ch1 GPIO 0x88
**TRACE [2016-12-22 13:15:52]: load profile 1
**TRACE [2016-12-22 13:15:54]: getSaveName
**TRACE [2016-12-22 13:15:54]: getSaveName 1
**TRACE [2016-12-22 13:15:54]: Saved at 
**TRACE [2016-12-22 13:15:55]: Saved at 2016-12-22 13:15:55
**TRACE [2016-12-22 13:15:55]: T01
**TRACE [2016-12-22 13:15:55]: T0.1: 0
**TRACE [2016-12-22 13:15:55]: T0.2: 0
**TRACE [2016-12-22 13:15:55]: T0.3
**TRACE [2016-12-22 13:15:55]: T0.4: 0
**TRACE [2016-12-22 13:15:55]: T0.5: 0
**TRACE [2016-12-22 13:15:55]: T0.6: 0
**TRACE [2016-12-22 13:15:55]: T0.7: 28
**TRACE [2016-12-22 13:15:55]: Saved at 2016-12-22 13:15:55
**TRACE [2016-12-22 13:15:55]: T0.8: 0
**TRACE [2016-12-22 13:15:55]: T2.7.1
**TRACE [2016-12-22 13:15:55]: T02
**TRACE [2016-12-22 13:16:11]: load profile 1
**TRACE [2016-12-22 13:16:12]: getSaveName
**TRACE [2016-12-22 13:16:12]: getSaveName 1
**TRACE [2016-12-22 13:16:12]: Saved at 
**TRACE [2016-12-22 13:16:12]: Saved at 2016-12-22 13:16:12
**TRACE [2016-12-22 13:16:12]: T01
**TRACE [2016-12-22 13:16:12]: T0.1: 0
**TRACE [2016-12-22 13:16:12]: T0.2: 0
**TRACE [2016-12-22 13:16:12]: T0.3
**TRACE [2016-12-22 13:16:12]: T0.4: 0
**TRACE [2016-12-22 13:16:13]: T0.5: 0
**TRACE [2016-12-22 13:16:13]: T0.6: 0
**TRACE [2016-12-22 13:16:13]: T0.7: 28
**TRACE [2016-12-22 13:16:13]: Saved at 2016-12-22 13:16:12
**TRACE [2016-12-22 13:16:13]: T0.8: 0
**TRACE [2016-12-22 13:16:13]: T2.7.1
**TRACE [2016-12-22 13:16:13]: T02
**TRACE [2016-12-22 13:16:31]: Ch1 GPIO 0x08
**TRACE [2016-12-22 13:16:32]: Ch1 GPIO 0x0a
**TRACE [2016-12-22 13:16:37]: load profile 1
**TRACE [2016-12-22 13:16:38]: getSaveName
**TRACE [2016-12-22 13:16:38]: getSaveName 1
**TRACE [2016-12-22 13:16:38]: Saved at 
**TRACE [2016-12-22 13:16:38]: Saved at 2016-12-22 13:16:38
**TRACE [2016-12-22 13:16:39]: T01
**TRACE [2016-12-22 13:16:39]: T0.1: 0
**TRACE [2016-12-22 13:16:39]: T0.2: 0
**TRACE [2016-12-22 13:16:39]: T0.3
**TRACE [2016-12-22 13:16:39]: T0.4: 0
**TRACE [2016-12-22 13:16:39]: T0.5: 0
**TRACE [2016-12-22 13:16:39]: T0.6: 0
**TRACE [2016-12-22 13:16:39]: T0.7: 28
**TRACE [2016-12-22 13:16:39]: Saved at 2016-12-22 13:16:38
**TRACE [2016-12-22 13:16:39]: T0.8: 0
**TRACE [2016-12-22 13:16:39]: T2.7.1
**TRACE [2016-12-22 13:16:39]: T02
**TRACE [2016-12-22 13:16:56]: Ch1 GPIO 0x8a
**TRACE [2016-12-22 13:16:56]: Ch1 GPIO 0x88
**TRACE [2016-12-22 13:17:04]: Ch1 GPIO 0x08
**TRACE [2016-12-22 13:17:04]: Ch1 GPIO 0x0a

mvladic, 2016-12-22:
Great! :) Thanks for helping us hunting this nasty bug. I removed all the trace lines and leaved just a bug fix. You can pull the latest version.

algaremi, 2016-12-22:
No problem, thanks to you guys. The save profile function is really a great tool. SB

algaremi, 2016-12-22:
BTW I just tried the 'git pull' but since I over-writed a lots of files, Git doesn't proceed unless I do a commit. I tried to find the answer but it is not clear.How can I force Git to download the latest version? Thanks
SB

mvladic, 2016-12-22:
You should discard all local changes before pull.  Try this one:

```
git checkout .
```

mvladic, 2016-12-22:
or:

```
git reset --hard
git pull
```

algaremi, 2016-12-22:
On 2016-12-22 13:46, Martin Vladic wrote:
> |git reset --hard git pull|

Tx

SB

prasimix, 2016-12-23:
Fixed

closed 2016-12-23 by prasimix

algaremi, 2016-12-30:
Hi. Just to let you know that I just updated the PSU firmware and I'm 
locked out of the user screen. I can see a 'Lock' icon on the bottom, 
but I cannot do anything else. Is it a new feature, and if yes how can I 
unlock it? If not, it is a bug I'm afraid.
If I press the lock, the buzzer bip, but nothing else...

mvladic, 2016-12-30:
You should be able to unlock if you press lock/unlock button for more then 2 seconds.

algaremi, 2016-12-30:
It works, Great. TxSB                                                                                                                                                                                                                                                                                                                                        Sent from my BlackBerry 10 smartphone on the Rogers network.                                                                                                                                                                                                                From: Martin VladicSent: Friday, December 30, 2016 3:47 AMTo: eez-open/psu-firmwareReply To: eez-open/psu-firmwareCc: algaremi; CommentSubject: Re: [eez-open/psu-firmware] Suggested user profile description when channels are coupled (#44)You should be able to unlock if you press lock/unlock button for more then 2 seconds.

algaremi, 2017-03-27:
Hi. Tried to compile the new M4 software version and got these errors:

sketch\ethernet.cpp: In function 'void eez::psu::ethernet::init()':

ethernet.cpp:143: error: 'class EthernetClass' has no member named 
'setDhcpTimeout'

      Ethernet.setDhcpTimeout(ETHERNET_DHCP_TIMEOUT * 1000UL);

               ^

Multiple libraries were found for "Ethernet2.h"
  Used: C:\Users\bergerons\Desktop\Documents\Arduino\libraries\Ethernet2
  Not used: C:\Program Files (x86)\Arduino\libraries\Ethernet2
Multiple libraries were found for "SD.h"
  Used: C:\Users\bergerons\Desktop\Documents\Arduino\libraries\SD
  Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
'class EthernetClass' has no member named 'setDhcpTimeout'

Any idea what's causing that?
Thanks
SB

mvladic, 2017-03-28:
Install `Ethernet2` library from `https://github.com/eez-open/Ethernet2` to your libraries folder.

algaremi, 2017-03-28:
Tx, SB                                                                                                                                                                                                                                                                                                                                        Sent from my BlackBerry 10 smartphone on the Rogers network.                                                                                                                                                                                                                From: Martin VladicSent: Tuesday, March 28, 2017 3:04 AMTo: eez-open/psu-firmwareReply To: eez-open/psu-firmwareCc: algaremi; CommentSubject: Re: [eez-open/psu-firmware] Suggested user profile description when channels are coupled (#44)Install Ethernet2 library from https://github.com/eez-open/Ethernet2 to your libraries folder.

algaremi, 2017-03-29:
Hi. I downloaded the Ethernet2 zip file, uncompressed the folder into my 
Arduino Libraires folder.
Still having an error, seem to be due to incorrect display type/config, 
see below.
any solution?
Thanks

Still have this error if I try to compile:

Arduino: 1.6.9 (Windows 7), Board: "Arduino Due (Programming Port)"

lcd.cpp:35: error: 'CTE32_R2' was not declared in this scope

  #define TFT_320QVT_9341 CTE32_R2

                          ^

sketch\conf_advanced.h:157:22: note: in expansion of macro 'TFT_320QVT_9341'

  #define DISPLAY_TYPE TFT_320QVT_9341

                       ^

sketch\lcd.cpp:37:14: note: in expansion of macro 'DISPLAY_TYPE'

  EEZ_UTFT lcd(DISPLAY_TYPE, LCD_RS, LCD_WR, LCD_CS, LCD_RESET);

               ^

Multiple libraries were found for "Ethernet2.h"
  Used: C:\Users\bergerons\Desktop\Documents\Arduino\libraries\Ethernet2
  Not used: C:\Program Files (x86)\Arduino\libraries\Ethernet2
Multiple libraries were found for "SD.h"
  Used: C:\Users\bergerons\Desktop\Documents\Arduino\libraries\SD
  Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
'CTE32_R2' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

'class EthernetClass' has no member named 'setDhcpTimeout'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

prasimix, 2017-03-29:
This is issue with TFT display library that like before reported `Ethernet2` is not related to M4 release, but to latest build. You can expect more "surprises" here since this is work in progress (related to e.g. #24). Therefore you can try to update to [M4 release](https://github.com/eez-open/psu-firmware/releases/tag/m4) or if you want to follow latest development download UTFT library from [here](http://www.rinkydinkelectronics.com/library.php?id=51). You probably has to remove from Arduino library folder `UTFT_Pro`.