eez-open/psu-firmware #59: Ethernet communication stops working in M3

prasimix, 2016-12-30:
Due to change in SPI transactions handling the Ethernet communication stops working.

prasimix, 2016-12-30:
When Ethernet is disabled with `#define OPTION_ETHERNET 0` it results with the following compilation error:
```
sketch/gui_page_sys_settings.cpp.o: In function `eez::psu::gui::SysSettingsEthernetPage::getData(eez::psu::gui::data::Cursor const&, unsigned char, eez::psu::gui::data::Snapshot*)':
sketch/gui_page_sys_settings.cpp:217: undefined reference to `eez::psu::ethernet::getIpAddress()'
sketch/gui_page_sys_settings.cpp:225: undefined reference to `eez::psu::ethernet::test_result'
sketch/psu.cpp.o: In function `psuReset':
sketch/psu.cpp:384: undefined reference to `eez::psu::ethernet::test_result'
sketch/psu.cpp:384: undefined reference to `eez::psu::ethernet::scpi_context'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Due (Programming Port).
```

mvladic, 2016-12-30:
Compilation error is fixed!

algaremi, 2017-01-03:
One issue with Ethernet no longer working if that if we activate the Ethernet, and restart the PSU, the unit stays locked out on the 'Waiting for Ehternet' screen. The only solution I found was to add these two lines:
#undef OPTION_ETHERNET
#define OPTION_ETHERNET 0
 into conf_user.h and upload the firmware again.

I don't know the code, but I would recommed using a time-out, let say 5 sec, when activating the Ethernet, and if the code doesn't find the DHCP server, the supply reverts back to operation without Ehternet.

prasimix, 2017-01-03:
We are aware of that, and due to issue with interrupts handling that cause us a lots of trouble before, now PSU blocks in delay function called in Ethernet initialization! Keep for a while Ethernet disable and we'll write about progress here.

algaremi, 2017-01-03:
Thanks for the update. I'l wait for the next development. SB

mvladic, 2017-01-04:
Fixed!

closed 2017-01-04 by prasimix