eez-open/psu-firmware #64: 10.2 Voltage and Current Calibration

Badger16438, 2017-01-09:
Following the Voltage and Calibration Guide:

INST CH1
OUTP 1
VOLT:PROT:STAT OFF
POW:PROT:STAT OFF
CURR:PROT:STAT OFF
CAL ON, "eezpsu"
CAL:VOLT:LEV MIN
CAL:VOLT 0.00
CAL:VOLT:LEV MID

When MID voltage level is selected, no output is measured from the power supply. If I issue a "CURR 0.5" to increase the current enough for the power supply to enter CV mode, voltage appears on the output.  I have not been able to calibrate the voltage correctly using this method though, resulting in 2.5v deviation from the selected voltage.

Admittedly this maybe my hardware, currently using the latest build a of firmware, with the r5B10 panel of PCB's.

prasimix, 2017-01-09:
When channel is in voltage calibration mode no additional CURR programming is needed. 
Also please note that _no load_ has to be connected! The MIDdle point should be somewhere about 19 V. When you say no output, does it mean something down to zero?
Please also let me know if you add in your conf_user.h the following (if you don't have TFT display installed):
``` arduino
#undef OPTION_DISPLAY
#define OPTION_DISPLAY 0
```

Badger16438, 2017-01-09:
No load connected. I shall test with display option of 0 now.

Badger16438, 2017-01-09:
With Option_Display set to 0 and the Power supply reflashed, the following was typed into the serial console:

INST CH1
OUTP 1
VOLT:PROT:STAT OFF
POW:PROT:STAT OFF
CURR:PROT:STAT OFF
CAL ON, "eezpsu"
CAL:VOLT:LEV MIN
CAL:VOLT 0.692
CAL:VOLT:LEV MID

The output voltage remains at 0.692v

if I end Calibration and type the following:

cal off, "eezpsu"
volt 5
curr 1

I end up with 2.97v output.

Volt 20 

I get 18.93

Volt 30 

I get 29.6v

I do note that I don't get any other LEDs lit up on the front panel other than the channel output LED. the CV or CC leds do not light.

Badger16438, 2017-01-09:
I performed the same Calibration example on CH 2 with the same results (slightly differing voltages - As to be expected).

prasimix, 2017-01-09:
Ok, so it seems that when uncalibrated you have expected output values with some tolerance (e.g. +/- 20-30 mV)?
Please let me know what values you are using for R50 and R60 on the Power board: 20K or 22K?
If nether CC or CV is active please check what R97 is 5K6 and R102 is 4K7 (that two values was mistakenly swapped in older release. They should be fine on r5B10 but double check. Do you have that problem only while in calibration mode or all the time?

Badger16438, 2017-01-09:
R50 & R60 = 223 =  22k
R97 = 5601 = 5k6
R102 = 4701 = 4k7

The LEDs have never lit, I have double checked polarity and soldering on the LED driver. I presume the LED driver is operating correctly, since the Output LEDs light correctly and no errors are reported. I assumed the LEDs were a separate issue.

Regarding the tolerance, as I approach 30v, yes i'm getting 29.6v which isn't too far away. Selecting 5v on the other hand gets me something approaching 3v.

Both channels are behaving identically, so i doubt its a soldering issue, I will however check each part against the BOM. The power board is a complex board and I guess its easy to place values incorrectly.

prasimix, 2017-01-09:
What you have as the input? You need to have min. 46VDC if you want to reach full 40V on the output

Badger16438, 2017-01-09:
Rectified voltage across the main filter caps is 49.5v

prasimix, 2017-01-09:
Ok, so you can go up to approx. 40V in that case (e.g. when send VOLT MAX)?

Badger16438, 2017-01-09:
No. only 38.9v Currently.

prasimix, 2017-01-09:
You can use this simple sketch to check if all LEDs and relays on the Arduino board are working. 
**Do not test it with Power board connected!**

[TLC5925_BP_TEST_r3B4.zip](https://github.com/eez-open/psu-firmware/files/694366/TLC5925_BP_TEST_r3B4.zip)

prasimix, 2017-01-09:
OK, 38.9V is a little bit below expected tolerance, but it's still strange that you cannot get MIDdle point while calibrating. Did you check all bias power: +/-5V for analog, +5V digital, and that +2.5V is on the voltage reference?

Badger16438, 2017-01-10:
Hi, Have checked all components positions and all looks good. The +/- 5V is ok, +5v Digital is good and the +2.5v Reference measured at C67 is good. The only difference between the original schematic and my build is the removal of R30, R33, Q6 current limiter components. I do notice that R32 next to Q4 is marked 2k2 on my schematic, however on R5B11 its been changed to 1k. I also note that JP9-0R on R5B10 connects R44 to PGND, but on R5B11 its pin 7 on IC7B to R44.

prasimix, 2017-01-10:
Oh, it seems that we know where is the problem :) Please move JP9-0R to the ground because that new offset feature that will allow us to go down to 0 when calibrated is still not supported into firmware!

prasimix, 2017-01-10:
R32 of 1K is fine, just leave it as is.

prasimix, 2017-01-10:
@mvladic: introduce new power board version (r5B10) on that board we have to take into consideration initial offset (50-100 mV)

@Badger16438: R70 could be increased from 5K1 to 10K since 50 mV offsets should be enough, but we'll lwt you know.

Badger16438, 2017-01-10:
I can confirm switching JP9 over and changing R32 to 1k 1% corrects the voltage issue. Many thanks for the LED / Relay test sketch, that Identified one LED - LED_SP, which wasn't soldered correctly.

Many thanks for your help :) Good luck with your crowdfunding - I'll be sure to pledge :)

prasimix, 2017-01-10:
Thanks to your valuable feedback too. You are actually ahead of us with revision, and that is a good reason for us to hurry with adding new board type and calibration into firmware ;)

Badger16438, 2017-01-10:
I look forward to your updates - I've backed your Crowd supply project for some more bare pcb's.

prasimix, 2017-02-09:
This can be now tested on the r5B10 Power board using the latest firmware build. To change board model please add the following code into your `conf_user.h `file:

``` arduino
#undef CHANNELS
#define CHANNELS \
    CHANNEL(1, CH_BOARD_REVISION_R5B10_PARAMS, CH_PINS_1, CH_PARAMS_50V_3A), \
    CHANNEL(2, CH_BOARD_REVISION_R5B10_PARAMS, CH_PINS_2, CH_PARAMS_50V_3A) \
```
After calibration both voltage and current could be set down to zero, but for the max level we'll need to change resistors values that will be reported here.
Also please take into account that channels will displays UR mode instead of CV or CC mode when set output values is low. That has to be fixed with changing some additional resistors to window comparators that will be reported too.

prasimix, 2017-02-10:
To fix CC/CV indications change R97, R102 to e.g. 10K. 
Increase R50, R60 to 24K to reach max. range of 40 V.

prasimix, 2017-02-11:
Calibration tested with r5B10 when reference GND is icreased to ~50 mV.

closed 2017-02-11 by prasimix