eez-open/psu-firmware #76: Touch screen calibration stuck.

dsaltas, 2017-01-25:
I am getting stuck on touch screen calibration. Touching yes or anything else just draws a point and I cannot continue. But if disable the external eeprom and reupload I don't have that issue. But again I have to calibrate the screen each time the psu starts. AT25256 is installed.
The DIAG:INFO:TEST? Returns the following among others "1, EEPROM, installed, passed"
I am having for now only ch#1 installed. Working on the second one.
Any ideas?
Thanks

prasimix, 2017-01-25:
Please send us serial monitor printout while calibration is in progress.

dsaltas, 2017-01-25:
EEZ PSU serial com ready
**TRACE: Touch modification ENABLED
**TRACE [2017-01-25 14:32:32]: PWR_DIRECT -> HIGH
**TRACE [2017-01-25 14:32:32]: Fan RPM: 4079
**TRACE [2017-01-25 14:32:41]: Calibration point: 3574, 432
**TRACE [2017-01-25 14:32:42]: Calibration point: 3576, 435
**TRACE [2017-01-25 14:32:42]: Calibration point: 3576, 435
**TRACE [2017-01-25 14:32:44]: Calibration point: 416, 3504
**TRACE [2017-01-25 14:32:44]: Calibration point: 416, 3506
**TRACE [2017-01-25 14:32:44]: Calibration point: 436, 3521
**TRACE [2017-01-25 14:32:44]: Calibration point: 436, 3521
**TRACE [2017-01-25 14:32:45]: Calibration point: 432, 435
**TRACE [2017-01-25 14:32:45]: Calibration point: 432, 440
**TRACE [2017-01-25 14:32:45]: Calibration point: 432, 440
**TRACE [2017-01-25 14:32:46]: Touch screen calibration points: 3576, 435, 436, 3521, 432, 440

and it stops there

mvladic, 2017-01-25:
Please, unzip following file and copy `touch_calibration.cpp` inside `eez_psu_sketch` folder:

[touch_calibration.zip](https://github.com/eez-open/psu-firmware/files/729671/touch_calibration.zip)

Test again and send us serial monitor printout.

dsaltas, 2017-01-25:
EEZ PSU serial com ready
**TRACE: Touch modification ENABLED
**TRACE [2017-01-25 14:50:09]: PWR_DIRECT -> HIGH
**TRACE [2017-01-25 14:50:10]: Fan RPM: 4107
**TRACE [2017-01-25 14:50:17]: Calibration point: 3518, 524
**TRACE [2017-01-25 14:50:17]: Calibration point: 3518, 524
**TRACE [2017-01-25 14:50:18]: Calibration point: 492, 3376
**TRACE [2017-01-25 14:50:18]: Calibration point: 506, 3380
**TRACE [2017-01-25 14:50:18]: Calibration point: 506, 3380
**TRACE [2017-01-25 14:50:19]: Calibration point: 480, 489
**TRACE [2017-01-25 14:50:19]: Calibration point: 480, 489
**TRACE [2017-01-25 14:50:19]: T1
**TRACE [2017-01-25 14:50:19]: Touch screen calibration points: 3518, 524, 506, 3380, 480, 489
**TRACE [2017-01-25 14:50:19]: T2
**TRACE [2017-01-25 14:50:19]: T3

mvladic, 2017-01-25:
Can you see "Save changes" page?

dsaltas, 2017-01-25:
Yes. Look.
[image]

mvladic, 2017-01-25:
With this change:

[gui.zip](https://github.com/eez-open/psu-firmware/files/729813/gui.zip)

please try the same thing - touching Yes while in "Save changes?" page.
Thanks!

dsaltas, 2017-01-25:
Can't compile. I got these errors.

sketch/gui.cpp: In function 'eez::psu::gui::Page* eez::psu::gui::createPageFromId(int)':
gui.cpp:136: error: 'PAGE_ID_CH_SETTINGS_ADV_VIEW' was not declared in this scope
     case PAGE_ID_CH_SETTINGS_ADV_VIEW: return new ChSettingsAdvViewPage();

          ^
gui.cpp:136: error: expected type-specifier before 'ChSettingsAdvViewPage'
     case PAGE_ID_CH_SETTINGS_ADV_VIEW: return new ChSettingsAdvViewPage();

                                                   ^
gui.cpp:136: error: expected ';' before 'ChSettingsAdvViewPage'
gui.cpp:136: error: 'ChSettingsAdvViewPage' was not declared in this scope
     case PAGE_ID_CH_SETTINGS_ADV_VIEW: return new ChSettingsAdvViewPage();

                                                                         ^
sketch/gui.cpp: In function 'void eez::psu::gui::showWelcomePage()':
gui.cpp:260: error: 'flush' was not declared in this scope
     flush();

           ^
sketch/gui.cpp: In function 'void eez::psu::gui::showSelfTestResultPage()':
gui.cpp:265: error: 'flush' was not declared in this scope
     flush();

           ^
sketch/gui.cpp: In function 'void eez::psu::gui::showStandbyPage()':
gui.cpp:270: error: 'flush' was not declared in this scope
     flush();

           ^
sketch/gui.cpp: In function 'void eez::psu::gui::showEnteringStandbyPage()':
gui.cpp:275: error: 'flush' was not declared in this scope
     flush();

           ^
sketch/gui.cpp: In function 'void eez::psu::gui::showEthernetInit()':
gui.cpp:280: error: 'flush' was not declared in this scope
     flush();

           ^
sketch/gui.cpp: At global scope:
gui.cpp:283: error: 'EnumItem' in namespace 'eez::psu::gui::data' does not name a type
 void pushSelectFromEnumPage(const data::EnumItem *enumDefinition, uint8_t currentValue, uint8_t disabledValue, void (*onSet)(uint8_t)) {

                                   ^
gui.cpp:283: error: ISO C++ forbids declaration of 'enumDefinition' with no type [-fpermissive]
 void pushSelectFromEnumPage(const data::EnumItem *enumDefinition, uint8_t currentValue, uint8_t disabledValue, void (*onSet)(uint8_t)) {

                                                   ^
sketch/gui.cpp: In function 'void eez::psu::gui::pushSelectFromEnumPage(const int*, uint8_t, uint8_t, void (*)(uint8_t))':
gui.cpp:284: error: expected type-specifier before 'SelectFromEnumPage'
     pushPage(INTERNAL_PAGE_ID_SELECT_FROM_ENUM, new SelectFromEnumPage(enumDefinition, currentValue, disabledValue, onSet));

                                                     ^
sketch/gui.cpp: In function 'void eez::psu::gui::changeLimit(const eez::psu::gui::data::Value&, float, float, float, void (*)(float))':
gui.cpp:364: error: 'NumericKeypadOptions' was not declared in this scope
  NumericKeypadOptions options;

  ^
gui.cpp:364: error: expected ';' before 'options'
  NumericKeypadOptions options;

                       ^
gui.cpp:366: error: 'options' was not declared in this scope
  options.editUnit = value.getType();

  ^
gui.cpp:378: error: 'NumericKeypad' has not been declared
  NumericKeypad::start(0, value, options, onSetLimit);
[... 50 lines cut ...]
                     selectWidget(g_foundWidgetAtDown);

                                                     ^
gui.cpp:867: error: 'deselectWidget' was not declared in this scope
                             deselectWidget();

                                            ^
gui.cpp:877: error: 'deselectWidget' was not declared in this scope
                             deselectWidget();

                                            ^
gui.cpp:924: error: 'deselectWidget' was not declared in this scope
                 deselectWidget();

                                ^
gui.cpp:963: error: 'PAGE_ID_ERROR_TOAST_ALERT' was not declared in this scope
     if (g_activePageId == PAGE_ID_TOAST3_ALERT || g_activePageId == PAGE_ID_ERROR_TOAST_ALERT) {

                                                                     ^
gui.cpp:972: error: 'drawTick' was not declared in this scope
         drawTick();

                  ^
exit status 1
'PAGE_ID_CH_SETTINGS_ADV_VIEW' was not declared in this scope

mvladic, 2017-01-25:
I think you don't have the latest version but M3 version. If true please try with this:

[gui.zip](https://github.com/eez-open/psu-firmware/files/729836/gui.zip)

Thanks!

dsaltas, 2017-01-25:
Yes, I have M3 version.

EEZ PSU serial com ready
**TRACE: Touch modification ENABLED
**TRACE [2017-01-25 15:56:23]: PWR_DIRECT -> HIGH
**TRACE [2017-01-25 15:56:23]: Fan RPM: 4079
**TRACE [2017-01-25 15:56:39]: Calibration point: 3555, 402
**TRACE [2017-01-25 15:56:39]: Calibration point: 3555, 402
**TRACE [2017-01-25 15:56:41]: Calibration point: 399, 3455
**TRACE [2017-01-25 15:56:41]: Calibration point: 399, 3451
**TRACE [2017-01-25 15:56:41]: Calibration point: 399, 3451
**TRACE [2017-01-25 15:56:43]: Calibration point: 399, 374
**TRACE [2017-01-25 15:56:43]: Calibration point: 399, 374
**TRACE [2017-01-25 15:56:43]: T1
**TRACE [2017-01-25 15:56:43]: Touch screen calibration points: 3555, 402, 399, 3451, 399, 374
**TRACE [2017-01-25 15:56:43]: T2
**TRACE [2017-01-25 15:56:43]: T3
**TRACE [2017-01-25 15:56:45]: 72, 180
**TRACE [2017-01-25 15:56:45]: 71, 180
**TRACE [2017-01-25 15:56:45]: 71, 181
**TRACE [2017-01-25 15:56:45]: 82, 180
**TRACE [2017-01-25 15:56:45]: 82, 180
**TRACE [2017-01-25 15:56:45]: 82, 180
**TRACE [2017-01-25 15:56:46]: 74, 180
**TRACE [2017-01-25 15:56:46]: 67, 181
**TRACE [2017-01-25 15:56:47]: 65, 176
**TRACE [2017-01-25 15:56:47]: 65, 176
**TRACE [2017-01-25 15:56:47]: 67, 181
**TRACE [2017-01-25 15:56:47]: 67, 181
**TRACE [2017-01-25 15:56:48]: 71, 181
**TRACE [2017-01-25 15:56:48]: 71, 181
**TRACE [2017-01-25 15:56:48]: 76, 178
**TRACE [2017-01-25 15:56:48]: 76, 178
**TRACE [2017-01-25 15:56:49]: 78, 181
**TRACE [2017-01-25 15:56:49]: 82, 179
**TRACE [2017-01-25 15:56:49]: 75, 178
**TRACE [2017-01-25 15:56:50]: 68, 181
**TRACE [2017-01-25 15:56:50]: 68, 177
**TRACE [2017-01-25 15:56:51]: 67, 183
**TRACE [2017-01-25 15:56:51]: 67, 183
**TRACE [2017-01-25 15:56:51]: 72, 183

mvladic, 2017-01-25:
Please,make the same test with this:

[gui.zip](https://github.com/eez-open/psu-firmware/files/729907/gui.zip)

Thanks!

dsaltas, 2017-01-25:
Here are the results.

EEZ PSU serial com ready
**TRACE: Touch modification ENABLED
**TRACE [2017-01-25 16:19:06]: PWR_DIRECT -> HIGH
**TRACE [2017-01-25 16:19:06]: Fan RPM: 4079
**TRACE [2017-01-25 16:19:12]: TG1
**TRACE [2017-01-25 16:19:13]: TG2
**TRACE [2017-01-25 16:19:14]: Calibration point: 3551, 409
**TRACE [2017-01-25 16:19:14]: Calibration point: 3551, 409
**TRACE [2017-01-25 16:19:15]: Calibration point: 395, 3518
**TRACE [2017-01-25 16:19:15]: Calibration point: 406, 3515
**TRACE [2017-01-25 16:19:15]: Calibration point: 406, 3515
**TRACE [2017-01-25 16:19:17]: Calibration point: 409, 438
**TRACE [2017-01-25 16:19:17]: Calibration point: 409, 438
**TRACE [2017-01-25 16:19:17]: T1
**TRACE [2017-01-25 16:19:17]: Touch screen calibration points: 3551, 409, 406, 3515, 409, 438
**TRACE [2017-01-25 16:19:17]: T2
**TRACE [2017-01-25 16:19:17]: T3
**TRACE [2017-01-25 16:19:21]: TG1
**TRACE [2017-01-25 16:19:21]: TG3
**TRACE [2017-01-25 16:19:21]: DOWN 72, 180
**TRACE [2017-01-25 16:19:21]: FWS 2188d
**TRACE [2017-01-25 16:19:21]: FWS 2202d
**TRACE [2017-01-25 16:19:21]: found
**TRACE [2017-01-25 16:19:21]: TG4
**TRACE [2017-01-25 16:19:21]: TG6
**TRACE [2017-01-25 16:19:21]: TG7
**TRACE [2017-01-25 16:19:21]: TG9
**TRACE [2017-01-25 16:19:21]: MOVE 72, 180
**TRACE [2017-01-25 16:19:21]: MOVE 72, 180
**TRACE [2017-01-25 16:19:21]: MOVE 69, 180
**TRACE [2017-01-25 16:19:21]: MOVE 67, 180
**TRACE [2017-01-25 16:19:22]: TG1
**TRACE [2017-01-25 16:19:22]: TG3
**TRACE [2017-01-25 16:19:22]: DOWN 79, 185
**TRACE [2017-01-25 16:19:22]: FWS 2188d
**TRACE [2017-01-25 16:19:22]: FWS 2202d
**TRACE [2017-01-25 16:19:22]: found
**TRACE [2017-01-25 16:19:22]: TG4
**TRACE [2017-01-25 16:19:22]: TG6
**TRACE [2017-01-25 16:19:22]: TG7
**TRACE [2017-01-25 16:19:22]: TG9
**TRACE [2017-01-25 16:19:22]: MOVE 79, 185
**TRACE [2017-01-25 16:19:22]: MOVE 79, 185
**TRACE [2017-01-25 16:19:22]: TG1
**TRACE [2017-01-25 16:19:23]: TG3
**TRACE [2017-01-25 16:19:23]: DOWN 63, 177
**TRACE [2017-01-25 16:19:23]: FWS 2188d
**TRACE [2017-01-25 16:19:23]: FWS 2202d
**TRACE [2017-01-25 16:19:23]: found
**TRACE [2017-01-25 16:19:23]: TG4
**TRACE [2017-01-25 16:19:23]: TG6
**TRACE [2017-01-25 16:19:23]: TG7
**TRACE [2017-01-25 16:19:23]: TG9
**TRACE [2017-01-25 16:19:23]: MOVE 63, 177
**TRACE [2017-01-25 16:19:23]: MOVE 63, 177
**TRACE [2017-01-25 16:19:23]: MOVE 72, 178
**TRACE [2017-01-25 16:19:23]: TG1
**TRACE [2017-01-25 16:19:23]: TG3
**TRACE [2017-01-25 16:19:23]: DOWN 83, 190
**TRACE [2017-01-25 16:19:23]: FWS 2188d
**TRACE [2017-01-25 16:19:23]: FWS 2202d
**TRACE [2017-01-25 16:19:23]: found
**TRACE [2017-01-25 16:19:23]: TG4
**TRACE [2017-01-25 16:19:24]: TG6
**TRACE [2017-01-25 16:19:24]: TG7
**TRACE [2017-01-25 16:19:24]: TG9
**TRACE [2017-01-25 16:19:24]: MOVE 83, 190
**TRACE [2017-01-25 16:19:24]: MOVE 83, 190
**TRACE [2017-01-25 16:19:24]: TG1
**TRACE [2017-01-25 16:19:24]: TG3
**TRACE [2017-01-25 16:19:24]: DOWN 61, 184
**TRACE [2017-01-25 16:19:24]: FWS 2188d
**TRACE [2017-01-25 16:19:24]: FWS 2202d
**TRACE [2017-01-25 16:19:24]: found
**TRACE [2017-01-25 16:19:24]: TG4
**TRACE [2017-01-25 16:19:24]: TG6
**TRACE [2017-01-25 16:19:24]: TG7
**TRACE [2017-01-25 16:19:24]: TG9
**TRACE [2017-01-25 16:19:24]: MOVE 61, 184
**TRACE [2017-01-25 16:19:24]: MOVE 61, 183
**TRACE [2017-01-25 16:19:25]: TG1
**TRACE [2017-01-25 16:19:25]: TG3
**TRACE [2017-01-25 16:19:25]: DOWN 57, 180
**TRACE [2017-01-25 16:19:25]: FWS 2188d
**TRACE [2017-01-25 16:19:25]: FWS 2202d
**TRACE [2017-01-25 16:19:25]: found
**TRACE [2017-01-25 16:19:25]: TG4
**TRACE [2017-01-25 16:19:25]: TG6
**TRACE [2017-01-25 16:19:25]: TG7
**TRACE [2017-01-25 16:19:25]: TG9
**TRACE [2017-01-25 16:19:25]: MOVE 57, 180
**TRACE [2017-01-25 16:19:25]: MOVE 57, 180
**TRACE [2017-01-25 16:19:26]: TG1
**TRACE [2017-01-25 16:19:26]: TG3
**TRACE [2017-01-25 16:19:26]: DOWN 61, 181
**TRACE [2017-01-25 16:19:26]: FWS 2188d
**TRACE [2017-01-25 16:19:26]: FWS 2202d
**TRACE [2017-01-25 16:19:26]: found
**TRACE [2017-01-25 16:19:26]: TG4
**TRACE [2017-01-25 16:19:26]: TG6
**TRACE [2017-01-25 16:19:26]: TG7
**TRACE [2017-01-25 16:19:26]: TG9
**TRACE [2017-01-25 16:19:26]: MOVE 61, 181

mvladic, 2017-01-25:
I think your device, due to the bug in our code, started with screen locked. With this change you should be able to finish calibration:

[eez_psu_sketch.zip](https://github.com/eez-open/psu-firmware/files/729956/eez_psu_sketch.zip)

But, as I said, screen will be locked so you need to press unlock button for 2 seconds to unlock the screen.

dsaltas, 2017-01-25:
I get an error while compiling.

gui.cpp:1973: error: macro "DebugTrace" passed 2 arguments, but takes just 1
             DebugTrace("TG4 %ud", found_widget.widgetOffset);

                                                            ^
sketch/gui.cpp: In function 'void eez::psu::gui::tick(long unsigned int)':
gui.cpp:1973: error: 'DebugTrace' was not declared in this scope
             DebugTrace("TG4 %ud", found_widget.widgetOffset);

             ^
exit status 1
macro "DebugTrace" passed 2 arguments, but takes just 1

Should I use latest version?

mvladic, 2017-01-25:
Ah, sorry! Please, try with this:

[eez_psu_sketch.zip](https://github.com/eez-open/psu-firmware/files/729980/eez_psu_sketch.zip)

mvladic, 2017-01-25:
> Should I use latest version?

The latest version also has this bug. We will probably make another M3 release with this bug fixed.

dsaltas, 2017-01-25:
It's working!!!

Many thanks!

mvladic, 2017-01-25:
Thank you for your great assistance in catching this bug!

prasimix, 2017-02-08:
Fixed

closed 2017-02-08 by prasimix