eez-open/psu-firmware #78: Suspicious file for antivirus

romain145, 2017-02-03:
The installer and the executable were treated as suspicious files by AVG Free.
I made a report saying that I trust both.
AVG answered and whitelisted the executable.

- Can someone else confirm that AVG is happy now?
- Other antivirus need to be checked, and recognise the file as safe.

electrokean, 2017-02-04:
I presume you're referring to the Simulator application for Windows?
I just submitted the installer to VirusTotal.com and it came back as clean on 54/55, except on Qihoo-360
https://www.virustotal.com/en/file/3736d23ed54e8c6a9b15332e5c9c70d049b6cd89f6a3ee6b8e239d7c193d5761/analysis/1486198048/

prasimix, 2017-03-04:
The latest analysis return [0/59](https://www.virustotal.com/en/file/3736d23ed54e8c6a9b15332e5c9c70d049b6cd89f6a3ee6b8e239d7c193d5761/analysis/). Therefore I'm going to close this issue.

closed 2017-03-04 by prasimix