eez-open/psu-firmware #81: Enhance firmware/software to provide features of a Source Measurement Unit

dagema, 2017-02-10:
Since the PS currently has both sourcing outputs and sensing inputs, it would seem possible to enhance the firmware/software to make it able to function as an SMU [https://en.wikipedia.org/wiki/Source_measure_unit](url)

An extremely simplified example of the concept can be seen here [http://www.changpuak.ch/electronics/Curve_Tracer.php](url) ,

and a professional modern product showing many of the possible capabilities can be seen here [http://www.mouser.com/ds/2/403/Keithley_2461DSL-794149.pdf](url)

In light of the above examples, the question is, what capabilities could be achieved with a reasonalbe amount of programming effort.  Perhaps there are some clever enthusiasts out there who could contribute and develop this as a great stretch goal.

prasimix, 2017-03-21:
Someone should make some draft specification first, taking into account PSU limitations.

dagema, 2017-03-22:
Firmware/software should offer a menue for selecting maximum voltage and maximum current.  After entering values the program should do a continuous sweep from 0 to Max voltage at a sweep frequency which the PSU is capable of supporting.  During each cycle of the sweep, the program should toggle between the voltage and current levels and show the results as a  plot on the display.  There should also be an option available to store a number of volatag/current configuations under a specific name which can be later loaded for testing differnt device categories.

The menue should also allow combining both outputs so that devices such as transistors can be measured, one channel for the device input (such as transistor base) and the other channel for the device output (such as transistor  emitter/collector).  In such cases, a selection between fixed or maiximum on either the input voltage channel or the output voltage channel should be required, and the sweep performed accordingly. Here 2 plots should be deplayed.

The software should automatically adjust the X/Y axis values according to the results and provide appropriate scale division values in the display.

dagema, 2017-03-22:
I had inadvertedly closed the issue. My mistake. (I thought the "Close and comment" button referred to closing the edit box.)

prasimix, 2017-03-22:
No problem, thanks for your input.

caizikun, 2017-12-08:
for the current range: 500mA, the resolution is only 0.1mA. I think this is hard to used as source meter due to resolution.
Mostly Source meter is used for optical LIV test. If current range is 0-500mA, the expected resolution is less than 0.03mA(14 bit DAC). For some laser, <10uA resolution are expected.

prasimix, 2017-12-08:
0.1 mA is currently used when it's used for power sourcing. We are still have 15-bit ADC for reading and 16-bit DAC for settings output value that can be used if eventually a sort of SMU functionality is going to be added. I said a sort, since it could be hardly in the same class as commercial device built with SMU functionality in mind.

prasimix, 2020-01-06:
SMU functionality will be developed within [EEZ BB3](https://github.com/eez-open/modular-psu) project. DCP405 modules already has 50 mA current range, and we'll try to make 2Q or 4Q power module, too.

closed 2020-01-06 by prasimix