eez-open/psu-firmware #82: Power supply freezes when controlling it over ethernet

dsaltas, 2017-02-15:
I was playing around controlling the psu over ethernet and noticed a few issues. Using telnet I managed to stay connected for half an hour with the psu and all my commands were executed flawlessly. I closed connection manually. No problems with that. Then I saw in the forum "eez-remote.py" a python script which controls the psu over ethernet. I ran it and although it managed to connect with the psu and read voltages and amps settings after a few seconds the psu beeped once and froze. I had to power cycle the psu to bring it back to life. I tried a few times again with the script and all my tries concluded to a frozen psu. Sometimes I manage to set a new voltage or ampere and switch the output on or off but again after a few seconds the script loses connection with the psu which freezes. Maybe there is a bug in ethernet library?

prasimix, 2017-02-26:
It could be caused by unresolved issue of using interrupts and SPI functions. Please check this once again with the most recent build since where polling instead of interrupt method is not used for data acquisition.

dsaltas, 2017-02-26:
I tried with the most recent build and I have the same problems. After a few seconds the psu freezes.

prasimix, 2017-02-27:
Thanks, we'll try to reproduce it on our side add let you know about status.

DwaineGarden, 2017-08-24:
I tried the latest firmware version with the eez-remote python example.   I'm unable to duplicate the freezing bug.    I have used both channels and turned on and off each channel.   I also updated the channels on the eez power supply itself when eez-remote is running.   eez-remote even picks up the current status when updating by the touchscreen.  I also have a load on channel one while testing...

[image]

Running Fedora 26 with Python3.2

DwaineGarden, 2017-08-24:
Here is the Python3 script example for reference.

https://github.com/kevinee/eez-remote

DwaineGarden, 2017-08-24:
I also increased the sample rate from every 2.5 seconds to 0.5 seconds.  Rock solid!  No PSU locking or freezing.

dsaltas, 2017-08-24:
I confirm that bug is gone in latest firmware.

My boards are "Arduino R3B4" and "Power R5B9".

closed 2017-08-24 by dsaltas