eez-open/psu-firmware #88: File manager for SD card

prasimix, 2017-03-03:
Add functionality to list, delete, modify, copy files stored on the SD card. The following type of files should be stored on the SD card:
- Copies of user profiles that resides on the EEPROM
- Values for programming voltage, current and duration of the `LIST`
- Data logging files  

`MMEMory` SCPI subsystem will be used for remotely managing SD card files.

prasimix, 2017-11-23:
At *RST, this value is not set to the root path.

Helix70, 2017-12-30:
Version 1.1 SD Card Failed on startup, but seems to respond to firmware anyway.
See SCPI commands below:

*IDN?
Envox,EEZ H24005 (Due),CS17054,v1.1

MMEM:CAT?
"System Volume Information,FOLD,0"

MMEM:INFO?
12288,7974543360

*TST?
1

DIAG:INFO:TEST?
"1, EEPROM, installed, passed","0, SD card, installed, failed","1, Ethernet, installed, passed","1, RTC, installed, passed","1, DateTime, installed, passed","2, BP option, installed, skipped","1, Fan, installed, passed","1, AUX temp, installed, passed","1, CH1 temp, installed, passed","1, CH2 temp, installed, passed","1, CH1 IOEXP, installed, passed","1, CH1 DAC, installed, passed","1, CH1 ADC, installed, passed","1, CH2 IOEXP, installed, passed","1, CH2 DAC, installed, passed","1, CH2 ADC, installed, passed"

Still had SD Card Fail on Power off/Power on/ Hard Reset, with a single high pitch beep - need to press home to continue.

Helix70, 2017-12-31:
If I move the lines

#if OPTION_SD_CARD
    sd_card::init();
#endif

down below

#if EEZ_PSU_SELECTED_REVISION == EEZ_PSU_REVISION_R3B4 || EEZ_PSU_SELECTED_REVISION == EEZ_PSU_REVISION_R5B12
    fan::init();
#endif

and recompiled, and that seems to avoid the error. Not sure if you needed the SD card earlier, but that was my observation. May be timing related.

mvladic, 2018-01-02:
Do you have this problem only on the latest version? We moved recently to a [different SD library](https://github.com/eez-open/SdFat).

Helix70, 2018-01-02:
Correct. No SD Card errors until 1.1 with the new SD card library.

TimWoodburn, 2018-03-06:
I've just updated to v1.1 and also have "SD Card Failed" message on start.

CPU R5B12, Arduino IDE 1.8.5
SD version 1.1.1
Used SDFAT from EEZ GIT

Fault was not apparent on V1.0

If I follow Helix70s steps I get the following:
**ERROR: -250, "Mass storage error"
**ERROR: -350, "Queue overflow"

test? shows "SD card, installed, failed"

Any ideas?

mvladic, 2018-03-07:
Please, upload latest firmware (we just committed some changes) and execute ```debug?``` SCPI command and let me know what you get. Thanks!

TimWoodburn, 2018-03-07:
Hi, 

The output is as follows:

> CH1 U_DAC = 1409
> CH2 U_DAC = 1409
> CH1 U_MON = 714
> CH2 U_MON = 672
> CH1 U_MON_DAC = 803
> CH2 U_MON_DAC = 690
> CH1 I_DAC = 1442
> CH2 I_DAC = 1442
> CH1 I_MON = 704
> CH2 I_MON = 702
> CH1 I_MON_DAC = 738
> CH2 I_MON_DAC = 738
> AUX TEMP = 330
> CH1 TEMP = 372
> CH2 TEMP = 371
> MAIN_LOOP_DURATION = 298 682 1148 / 200 623 1154 / 158 3485594
> ADC_COUNTER = 0 / 0 / 20
> Dynamic ram used: 584
> Program static ram used 33568
> Stack ram used 4360
> My guess at free mem: 59792
> SD volume is FAT 32
> SD card begin result: 1
> SD fs begin result: 1
> SD blocks per cluster: 8
> SD cluster count: 1945891
> SD fat start block: 8226
> SD fat count: 2
> SD blocks per fat: 15203
> SD root dir start: 2
> SD data start block: 38632
> SD card size: 7990 MB
> SD flash erase size: 128 blocks
> SD erase single block
> SD data area is not aligned on flash erase boundaries! Download and use formatter from www.sdcard.org!

mvladic, 2018-03-07:
Hm, strange. "card begin result" and "fs begin result" is 1, which is good and you shouldn't get "SD card, installed, failed".

TimWoodburn, 2018-03-07:
This unit was only assembled a couple of weeks ago however I don't recall noticing the SD card error with v1.0.

I pushed the SD card and rebooted...failed on start. Rebooted again, fault no longer apparent.  I've recycled the power a few times and _mostly_ no failures.

Perhaps the card wasn't fully located...seems odd though that the fault wasn't apparent on V1.0.

Thanks for having a look.

mvladic, 2018-03-08:
Would you be so kind and next time when you get SD failure, execute "debug?" and send me (we should see "SD card begin result error: ???" line with SdFat library internal error code).

mvladic, 2018-04-06:
Because of:

> SD fs begin result: 0

I know that filesystem initialization is failing. But, I need more info.

Please try the same thing with the latest [psu-firmware](https://github.com/eez-open/psu-firmware) and [SdFat library](https://github.com/eez-open/SdFat). 

Hopefully we will se after `SD fs begin result: 0` one more line with additional info. Thanks!

Helix70, 2018-04-06:
I updated and tried the master firmware.
It seems ok. Now I get this from debug?

CH1 U_DAC = 23372
CH2 U_DAC = 20550
CH1 U_MON = 633
CH2 U_MON = 653
CH1 U_MON_DAC = 11623
CH2 U_MON_DAC = 10207
CH1 I_DAC = 51830

CH2 I_DAC = 14084
CH1 I_MON = 683
CH2 I_MON = 707
CH1 I_MON_DAC = 25871
CH2 I_MON_DAC = 7020
AUX TEMP = 340
CH1 TEMP = 371
CH2 TEMP = 368
MAIN_LOOP_DURATION = 206 5500 10808 / 206 5501 11641 / 204 4101765
ADC_COUNTER = 0 / 0 / 12
Dynamic ram used: 436
Program static ram used 34116
Stack ram used 4392
My guess at free mem: 59360
SD volume is FAT 32
SD card begin result: 1
SD fs begin result: 1
SD blocks per cluster: 8
SD cluster count: 1946913
SD fat start block: 34
SD fat count: 2
SD blocks per fat: 15211
SD root dir start: 2
SD data start block: 30456
SD card size: 7990 MB
SD flash erase size: 128 blocks
SD erase single block
SD data area is not aligned on flash erase boundaries! Download and use formatter from www.sdcard.org!

mvladic, 2018-04-06:
I'm more interested in debug info when SD card initialization fails ;)

closed 2018-04-29 by prasimix