eez-open/psu-firmware #95: Examples of connection with a SCPI contoller

prasimix, 2017-03-19:
Idea is to generate over time a collection of various examples how to communicate with firmware via 3rd party so-called SCPI controller software. That could be, if applicable, LabVIEW, BenchVue, MyOpenLab, etc. or your own application.

dagema, 2017-08-30:
I have been trying to use the SCPI interface.  I can connect to the EEZ PSU via TCP/IP using Keysight Command Expert with Generic SCPI-99 Instrument / 1.0 command set, and some commands work while others seem to be missing.  Examples of working commands are: SYSTem:ERRor?, SYSTem:VERSion?,
STATus:PRESet, STATus:OPERation:ENABle?, STATus:OPERation:EVENt?, STATus:QUEStionable:ENABle?, STATus:QUEStionable:ENABle{integer}, *CLS, *IDN?, *OPC?, *SRE{integer}, *SRE?, *STB? and *TST?.  So I know I have a working connection.  However, I do not find any power supply commands such as SOURce:VOLTage:LEVel, SOURce:CURRent:LEVel etc. as I would expect for a power supply. I would like to begin to get this collection of examples started, but I have not been able to reconcile the examples provided in the EEZ PSU SCPI reference v1.0 document.  Can you provide ANY working example of a SCPI  command file which you have most certainly used to test the EEZ H24005 SCPI firmware implementation?  If I can get a short jump-start, that would enable me to develop some usefull examples.

Thank you.

prasimix, 2017-08-30:
Hm, I'm not sure to which extend Keysight Command Expert is usable (at least some of recent versions such as 2017). It don't allow you to send anything what is not specified under command set of selected instrument, and _Generic SCPI-99 Instrument / 1.0 command set_ is laughably limited. For that we should have specification of its "instrument" or maybe with VISA driver that could be possible. But someone has to build VISA driver (I believe that some kind of "developer" LabVIEW version exists that allows such thing by relatively simple listing all supported SCPI commands).

dagema, 2017-08-30:
OK, I have both LabVIEW and Matlab.  I will give them a try.
Thanks for the input.

dagema, 2017-10-28:
I am unable to generate a VISA driver using LabVIEW, because the device is not compatible with VISA with a null value for the serial number.  Is there a way that I can set the H24005 serial number?

prasimix, 2017-10-28:
You can check serial number with SYST:SER? and set it with same command. It's a string up to 7 character long, e.g. SYST:SER "PSU0001"