eez-open/psu-firmware #135: Inhibit Arduino Due reset when serial connection is established

prasimix, 2017-05-27:
Arduino Due will be reset every time when serial (via USB) connection is established. That is possibly a feature needed for new firmware upload (using Arduino IDE) but it's unnecessary for normal operation.

jlegen, 2017-06-12:
forgive my potential ignorance, but what about using the **native** USB port for serial communications? If **not** opened at 1200bps, there will be no reset, and the **programming** USB port might still be used for regular firmware uploads...

prasimix, 2017-06-12:
Forgive my ignorance too :) I'm not aware of such potential difference in behavior between native and programming port. The thing is that only one USB port was made exposed (and isolated) on rear panel and if we want to replace it with native one, for every new firmware uploading user has to open top cover and move internal USB cable or connect another cable directly to the programming port.

jlegen, 2017-06-12:
Why should a user have to open the top cover?😎 You probably did not use Arduino IDE that much yet - if you choose the Due board in IDE, you'll find **two** boards named 'Due with native port' and 'Due with programming port'. Then the IDE will use the chosen one for the upload...😊
Difference in fact is the upload-handling of the IDE: the prog port - provided by Due's additional Atmega16 - does the automatic reset, the native port has to be triggered for the reset by the IDE upload procedure ('magic bit rate').
For the program code, a different USB library has to be used (afair 'usbserial' or 'serial')

prasimix, 2017-06-14:
Thanks, I'm aware of possiblity to choose between two USB ports for firmware upload. I tested another thing once again: does native USB can be used as serial monitor port or not. It seems that is not possible with "default" settings (whatever that mean).

jlegen, 2017-06-14:
"default settings" probably means, that e.g. the default examples (and standard/common code) won't work, unless you replace "Serial.begin", "Serial.println" with "SerialUSB.begin", "SerialUSB.println" etc.
This *should* work including serial monitor, though i have to admit that i never used that myself. Only played with uploads so far. While typing this - one oddity exists because it is "native USB" - after each firmware upload, the USB connection is lost (resp. re-established by newer IDEs), because USB-mimic resets itself... This might be a reason why serial monitor might fail after an upload if not used with Arduino IDE.
The good thing: native port can be used as "host" - so we might plug a mouse or keyboard into EEZ PSU... :-P

prasimix, 2017-07-18:
Fixed

closed 2017-07-18 by prasimix