eez-open/psu-firmware #149: Calibration issues

dsaltas, 2017-06-25:
* Firmware version: 1.0 and M5
* Shield version: R3B4
* Powerboard version: R5B9

### Expected behavior
After successfully calibration I was expecting that I can set voltage and amperage to any value I want in the range 0-40V and 0-5A.

### Actual behavior
I can only set voltage and amperage in very limited ranges.
For voltage: 36.66V-36.74V and for amperage: 4.64A-4.65A.

prasimix, 2017-06-25:
This issue is fixed for Power boards r5B12 in 4ae91bc but no testing was performed on older boards.

mvladic, 2017-06-27:
Please try it now.

dsaltas, 2017-06-27:
Well done! It works now, but I discovered another bug.
If I press Settings from Ch#2 and then select either Current/Voltage limit and OCP/OVP it goes to Ch#1.

mvladic, 2017-06-28:
> If I press Settings from Ch#2 and then select either Current/Voltage limit and OCP/OVP it goes to Ch#1.

Fixed!

closed 2017-06-28 by dsaltas