eez-open/psu-firmware #157: Compile issue in 1.0.1?

amuir2k, 2017-08-19:
I had to comment out the following line in ethernet.cpp in the 1.01 (and master) builds to get the project to compile, 
 Ethernet.setDhcpTimeout(ETHERNET_DHCP_TIMEOUT * 1000UL);

Leaving it in yields the following compile error:

sketch/ethernet.cpp: In function 'void eez::psu::ethernet::init()':
ethernet.cpp:146: error: 'class EthernetClass' has no member named 'setDhcpTimeout'
     Ethernet.setDhcpTimeout(ETHERNET_DHCP_TIMEOUT * 1000UL);
              ^
exit status 1
'class EthernetClass' has no member named 'setDhcpTimeout'

I've updated all the IDE libraries, but no help.  Is this a bug or did I misconfigure something?
Thanks

Art

mvladic, 2017-08-19:
Use following Ethernet2 library:

https://github.com/eez-open/Ethernet2/commits/master

amuir2k, 2017-08-19:
Thanks, worked like a charm!

closed 2017-08-19 by prasimix