eez-open/psu-firmware #158: Feature Request: Support for battery charking profiles

sorlok72, 2017-08-20:
Would like to see charging of  different battery types, sizes and charging power.
Like LiPo, NiMH, LiFe, etc.  Or perhaps make it with profile, no new battery types can be supported 
without new firmware needed.?
Thanks for a great product

prasimix, 2017-08-20:
Thanks for that. It will be nice if you or someone else can provide instructions for charging various battery types. In the mean time I hope you can manage it by using simple list with just one or two steps:
1. Charging time with max. charging voltage and charging current
2. "float" time with voltage limit and small current
If second step is not needed you can define that channel output is disabled on the list end.

@mvladic: this is a first good case for implementing "program" infrastructure. Such program should be uploaded from EEZ Studio and include instructions and own set of GUI pages for displaying and entering program parameters, etc. If we provide that "program" can also store and retrieve data from SD card (what could be beneficial for battery models, charge logging, etc.) then we should first to address #88.

DwaineGarden, 2017-08-24:
+1.   I would love to see battery profiles based of the data sheets specs for different batteries.  Even up to a vehicle 12volt battery.

sorlok72, 2017-08-30:
Alter researching a bit on the subject, there are 3 ways to charge LIPO, LiFo etc.
Balanced charging, individual measuring and charging each cell
Charging all celle at once, monitoring individual cells
Just charging all cell at once, not nice/safe

So to do this would require allot of channels and sens lines.
I dont think this is doable without HW changes,

prasimix, 2017-08-30:
Thanks for comment, but this issue is still valid for some older type of batteries like Pb, NiCd, NiHm or that is not interesting nowadays to anyone?

sorlok72, 2017-08-30:
No problem, just wanted to stop the part I created, so you would not waste
time on it

momo-aux, 2018-02-28:
To load a NiMh battery there should be some kind of trigger that can be used to detect a -5mV voltage drop and if that occurs stop the charging process.

So would it be possible to have a programm in the firmware that I could activate to make this happen?

Like: Set the PSU to 11.6V with 150mA, connect the 9.6V NiMh 1500mAh Battery pack and start charging. If the Voltage raises all is ok. If Voltage starts dropping after some time, the NiMh Battery is fully charged and the PSU should power off.

prasimix, 2018-02-28:
Currently it's not possible to do that in firmware, i.e. without external agency, that is PC connected via USB or Ethernet. In that case it's possible to write a simple script that will measure voltage change tendency and stop the charging process when mentioned criteria is met. We are currently working on EEZ Studio that has SCPI console that can be used for communication using SCPI commands. In next step we'll add "macros" or simple language such as Blockly, etc. for tasks like one that you asked for.

momo-aux, 2018-02-28:
Great news, thank you.