eez-open/psu-firmware #159: Compile problem in 1.0.1 with Ethernet2 library

ExplodingLemur, 2017-08-21:
When compiling 1.0.1 using the Ethernet2 library from either the Arduino repositories or from https://github.com/eez-open/Ethernet2, I receive this error:

````
Arduino: 1.8.2 (Windows 10), TD: 1.36, Board: "Arduino Due (Native USB Port)"

sketch\ethernet.cpp: In function 'void eez::psu::ethernet::init()':

ethernet.cpp:146: error: 'class EthernetClass' has no member named 'setDhcpTimeout'

     Ethernet.setDhcpTimeout(ETHERNET_DHCP_TIMEOUT * 1000UL);

              ^

exit status 1
'class EthernetClass' has no member named 'setDhcpTimeout'
````

The full verbose compile log is attached.
[eez-1.0.1-error.txt](https://github.com/eez-open/psu-firmware/files/1237501/eez-1.0.1-error.txt)

This appears to be a duplicate of #157 however it is not resolved by using the Ethernet2 library from the eez-open repository.

prasimix, 2017-08-21:
Maybe you should move that library from `C:\Users\ExplodingLemur\Documents\Arduino\libraries\Ethernet2-1.0.2`  to `C:\Users\ExplodingLemur\Documents\Arduino\libraries\Ethernet2`

mvladic, 2017-08-21:
Are you sure Arduino library is removed from the libraries folder?

ExplodingLemur, 2017-08-23:
prasimix: Renaming the directory doesn't change the result.
mvladic: Yes, there is no Ethernet2 library in the Arduino libraries directory, only the eez-open version in my local libraries directory.

mvladic, 2017-08-23:
The name of directory doesn't matter. 

Go to your `C:\Users\ExplodingLemur\Documents\Arduino\libraries\Ethernet2-1.0.2\src` directory and open file Ethernet2.h. It should look like this (i.e. function setDhcpTimeout should be present inside EthernetClass.):

```
...
class EthernetClass {
private:
  IPAddress _dnsServerAddress;
  char* _dnsDomainName;
  char* _hostName;
  DhcpClass* _dhcp;
  unsigned long _dhcpTimeout;
public:
  uint8_t w5500_cspin;

  static uint8_t _state[MAX_SOCK_NUM];
  static uint16_t _server_port[MAX_SOCK_NUM];

  EthernetClass() { _dhcp = NULL; w5500_cspin = 10; _dhcpTimeout = 60000; }
  void init(uint8_t _cspin = 10) { w5500_cspin = _cspin; }

  void setDhcpTimeout(unsigned long timeout) {
    _dhcpTimeout = timeout;
  }
...
```
If that is not the case then you don't have eez-open version of Ehternet2 library.

ExplodingLemur, 2017-08-24:
I grabbed 1.0.2 from https://github.com/eez-open/Ethernet2/releases
The Ethernet2.h file only has the IPAddress and DhcpClass* private methods:
```
class EthernetClass {
private:
  IPAddress _dnsServerAddress;
  DhcpClass* _dhcp;
public:
  uint8_t w5500_cspin;

  static uint8_t _state[MAX_SOCK_NUM];
  static uint16_t _server_port[MAX_SOCK_NUM];

  EthernetClass() { w5500_cspin = 10; }
  void init(uint8_t _cspin = 10) { w5500_cspin = _cspin; }
...
```

ExplodingLemur, 2017-08-24:
I just checked and the 1.0.1 and 1.0 releases from that repository don't have those changes either.

G33KatWork, 2017-08-24:
I just ran into the same problem. Don't use the release but the master branch of the EEZ-PSU Ethernet2 library.

mvladic, 2017-08-24:
> Don't use the release but the master branch of the EEZ-PSU Ethernet2 library.

Exactly! Latest release was before I forked Ethernet2 library. To prevent future mistakes I just made a new release with the latest version from master.

DwaineGarden, 2017-08-24:
I used the master and compiled the firmware without any errors under v1.8.3 and nightly build of the Arduino IDE.

dagema, 2017-08-25:
> Exactly! Latest release was before I forked Ethernet2 library. To prevent future mistakes I just made a new release with the latest version from master.

I am confused.  I just performed a download from the "Latest release"  - /url) /releases[https://github.com/eez-open/psu-firmware/releases](/url) - Source code (zip) -

and still have the same problem - no ethernet2.h

What am I missing?

prasimix, 2017-08-25:
Did you download the latest Ethernet2 lib from [here](https://github.com/eez-open/Ethernet2/releases/tag/1.0.2-eez)?

dagema, 2017-08-25:
No. That node does not contain any arduino.ino project.  How do I integrate that node into the "Latest release" for Arduino sketch?

prasimix, 2017-08-25:
Libraries has to be placed into separate folder as noted in the firmware upload instructions [here.](http://www.envox.hr/eez/bench-power-supply/building-instructions.html#instr_fw)

dagema, 2017-08-25:
I copied subfolder "src" to the library subfolder "ethernet", and now get the following errors:

<pre>
<font color="#95a5a6">/**</font>
<font color="#95a5a6"> * EEZ PSU Firmware</font>
<font color="#95a5a6"> * Copyright (C) 2015 Envox d.o.o.</font>
<font color="#95a5a6"> *</font>
<font color="#95a5a6"> * This program is free software: you can redistribute it and/or modify</font>
<font color="#95a5a6"> * it under the terms of the GNU General Public License as published by</font>
<font color="#95a5a6"> * the Free Software Foundation, either version 3 of the License, or</font>
<font color="#95a5a6"> * (at your option) any later version.</font>
<font color="#95a5a6"></font>
<font color="#95a5a6"> * This program is distributed in the hope that it will be useful,</font>
<font color="#95a5a6"> * but WITHOUT ANY WARRANTY; without even the implied warranty of</font>
<font color="#95a5a6"> * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. &nbsp;See the</font>
<font color="#95a5a6"> * GNU General Public License for more details.</font>
<font color="#95a5a6"></font>
<font color="#95a5a6"> * You should have received a copy of the GNU General Public License</font>
<font color="#95a5a6"> * along with this program. &nbsp;If not, see &lt;</font><u><font color="#95a5a6">http://www.gnu.org/licenses/</font></u><font color="#95a5a6">&gt;.</font>
<font color="#95a5a6"> */</font>
 
<font color="#5e6d03">#include</font> <font color="#005c5f">"SPI.h"</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">scpi</font><font color="#434f54">-</font><font color="#000000">parser</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#005c5f">"psu.h"</font>

<font color="#5e6d03">#if</font> <font color="#000000">OPTION_ETHERNET</font>
<font color="#5e6d03">#if</font> <font color="#000000">EEZ_PSU_SELECTED_REVISION</font> <font color="#434f54">==</font> <font color="#000000">EEZ_PSU_REVISION_R1B9</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">UIPEthernet</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">UIPServer</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">UIPClient</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#elif</font> <font color="#000000">EEZ_PSU_SELECTED_REVISION</font> <font color="#434f54">==</font> <font color="#000000">EEZ_PSU_REVISION_R3B4</font> <font color="#434f54">||</font> <font color="#000000">EEZ_PSU_SELECTED_REVISION</font> <font color="#434f54">==</font> <font color="#000000">EEZ_PSU_REVISION_R5B12</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Ethernet2</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><b><font color="#d35400">EthernetServer</font></b><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><b><font color="#d35400">EthernetClient</font></b><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#endif</font>
<font color="#5e6d03">#endif</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;<font color="#000000">PSU_boot</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;<font color="#000000">PSU_tick</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

</pre>

prasimix, 2017-08-25:
What you copied is not an error text.

dagema, 2017-08-25:
Please look again. I highlighted the errors with bold type.

mvladic, 2017-08-25:
> No. That node does not contain any arduino.ino project. How do I integrate that node into the "Latest release" for Arduino sketch?

Arduino libraries (such as Ethernet2) doesn't contain aruduino.ino project because you don't need to compile libary as standalone project. 

You need to compile [psu-firmware](https://github.com/eez-open/psu-firmware) - that's why you are here. To compile psu-firmware, among else, you must place [Ethernet2 library](https://github.com/eez-open/Ethernet2) inside Arduino libraries folder.

dagema, 2017-08-25:
Sorry, I am not very proficient in Arduino usage.  Do I have to rename the ethernet2-master directory to ethernet2, or do I just copy the directory as downloaded into the Arduino library?

mvladic, 2017-08-25:
You don't need to rename it, just copy the directory.

dagema, 2017-08-25:
I copied the directory into the Arduino library directory.  I now get following error:

`C:\Users\User\Downloads\psu-firmware-1.01\psu-firmware-1.01\eez_psu_sketch\eez_psu_sketch.ino:31:23: fatal error: Ethernet2.h: No such file or directory

 #include <Ethernet2.h>

                       ^

compilation terminated.

Using library SPI at version 1.0 in folder: C:\ProgramData\MATLAB\SupportPackages\R2017a\3P.instrset\arduinoide.instrset\idepkgs\packages\arduino\hardware\sam\1.6.7\libraries\SPI 
Using library scpi-parser at version 1.0.0 in folder: C:\Program Files (x86)\Arduino\libraries\scpi-parser 
Using library eez_psu_lib at version 1.0.0 in folder: C:\Program Files (x86)\Arduino\libraries\eez_psu_lib 
exit status 1
Error compiling for board Arduino Due (Native USB Port).
Invalid library found in C:\Program Files (x86)\Arduino\libraries\Ethernet2-master: C:\Program Files (x86)\Arduino\libraries\Ethernet2-master
`

mvladic, 2017-08-25:
What folders and files are inside C:\Program Files (x86)\Arduino\libraries\Ethernet2-master?

dagema, 2017-08-25:
All the files/directories found here: https://github.com/eez-open/Ethernet2

mvladic, 2017-08-25:
Could you post complete dump of compiling (make sure 'Show verbose output during compilation' in options is enabled)?

dagema, 2017-08-25:
Sorry, but the editor says my comilation dump is more than 65535 charachters long and will not accept it.

dagema, 2017-08-26:
The problem is now solved.  I had been copying to the Arduino installation library directory instead of the Arduino user space library directory, which was the mistake.  After copying both library including ethernet2 into the Arduino user space library directory, compilation performs without error.

Thank yoiu for a great product!

dagema, 2017-08-26:
I was too quick to rejoice: after successful upload of new firmware, the PS does not come out of standby mode. Here is the upload protocol: 
`Linking everything together...
"C:\ProgramData\MATLAB\SupportPackages\R2017a\3P.instrset\arduinoide.instrset\idepkgs\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-gcc" -mcpu=cortex-m3 -mthumb -Os -Wl,--gc-sections "-TC:\ProgramData\MATLAB\SupportPackages\R2017a\3P.instrset\arduinoide.instrset\idepkgs\packages\arduino\hardware\sam\1.6.7\variants\arduino_due_x/linker_scripts/gcc/flash.ld" "-Wl,-Map,C:\Users\User\AppData\Local\Temp\arduino_build_362085/eez_psu_sketch.ino.map"  -o "C:\Users\User\AppData\Local\Temp\arduino_build_362085/eez_psu_sketch.ino.elf" "-LC:\Users\User\AppData\Local\Temp\arduino_build_362085" -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--start-group "C:\Users\User\AppData\Local\Temp\arduino_build_362085/core/syscalls_sam3.c.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\actions.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\adc.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\arduino_util.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\board.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\bp.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\buzzer.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\calibration.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\channel.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\channel_dispatcher.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\dac.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\datetime.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\debug.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\devices.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\eeprom.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\eez_psu_sketch.ino.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\encoder.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\ethernet.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\event_queue.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\fan.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\font.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gesture.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_calibration.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_data.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_document.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_edit_mode.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_edit_mode_keypad.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_edit_mode_slider.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_edit_mode_step.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_keypad.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_numeric_keypad.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_page.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_page_ch_settings_adv.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_page_ch_settings_info.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_page_ch_settings_protection.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_page_ch_settings_trigger.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_page_event_queue.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_page_main.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_page_self_test_result.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_page_sys_info.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_page_sys_settings.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_page_user_profiles.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_password.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_view.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\gui_widget_button_group.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\io_pins.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\ioexp.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\lcd.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\list.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\ntp.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\ontime.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\persist_conf.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\profile.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sketch\psu.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_362085\sket [... cut ...]
[===========                   ] 39% (656/1644 pages)
[==============                ] 49% (820/1644 pages)
[=================             ] 59% (984/1644 pages)
[====================          ] 69% (1148/1644 pages)
[=======================       ] 79% (1312/1644 pages)
[==========================    ] 89% (1476/1644 pages)
[============================= ] 99% (1640/1644 pages)
[==============================] 100% (1644/1644 pages)
Verify successful
done in 74.092 seconds
Set boot flash true
CPU reset.
`
Can you help, please?

prasimix, 2017-08-26:
First, 80 seconds upload (8x slower) indicate that you are using programming instead of native USB port. That is fine as long as you define that in your conf_user.h  

Did you try to reset Arduino after upload? Also if your latest state was standby, that state will remain after reset. Did you try to touch the screen to wake it up from standby?

dagema, 2017-08-26:
Yes, I tried all, but no reaction.  I now performed a fresh installation of Windows 7 Profressional on a new laptop and installed new Arduino ISE 1.8.4 on that laptop. After connection the PSU to that laptop I was able to install the new firmware and the PSU started normally and shows Firmware v1.01.  Problem is therefore now solved.  Apparantly there is a compatability issue with some additional Arduino related software which I have installed, presumably the MATLAB installation.

Thanks you for your support.

ExplodingLemur, 2017-08-26:
Thanks.  Hadn't realized the releases were behind master.  It builds just fine now.

closed 2017-08-26 by ExplodingLemur

prasimix, 2017-08-28:
A short video about installing Arduino IDE and firmware components is now available [here](https://www.youtube.com/watch?v=Ar5kHiepsNM).