eez-open/psu-firmware #160: Unable to complete the calibration process

Len42, 2017-08-26:
Firmware version 1.01 (also observed on 1.0)
Shield & Powerboard version R5B12

When I perform the calibration process on either channel, I cannot get past the mid-current setting. It will not accept the number that I input, giving the error "Value out of range". I have tried entering the value in amps ("2.478A") or milliamps ("2478ma") and also tried some other random values but it always gives that error.

Here is the debug output from the serial port during calibration (The measured current in this case was 2.478 A):

EEZ,PSU 2/40/05 (Due),CS17296,v1.01
**TRACE [2017-08-26 17:29:10]: Voltage range: 0.000000 - 40.000000V
**TRACE [2017-08-26 17:29:10]: ADC=0.200779
**TRACE [2017-08-26 17:29:57]: ADC check failed: level=2.425000, adc=-0.002270, diff=2.427270, allowedDiff=0.475000

prasimix, 2017-08-26:
Ok, please let me know what is output voltage during current calibration _without_ load (should be 20 V) and _with_ connected load. I presume that you are using appropriate load (e.g. 4 Ohm or smaller)

DwaineGarden, 2017-08-27:
I have the same problem.  Looks like when calibrating in current mode the psu is in CV.  I have 20v at 2484ma with each current test.  

Once I got the psu to drop to CC on the 50ma test. I enter the value measured and it accepted it. Next current test back to only CV with 2484ma

Using a Keysight U1272a meter and a 1ohm 25watt @1 % resister

DwaineGarden, 2017-08-27:
Here are some photos of the situation during calibration.  

[image]
[image]
[image]

prasimix, 2017-08-27:
Ok, can you measure output current of e.g. about 1A when you set that current outside calibration wizard?

DwaineGarden, 2017-08-27:
Here is the test results.  

[image]

prasimix, 2017-08-27:
Ok, as you can get 1A you should get 2.5 or max. of about 5 A. All what I can say that your load is not connected during the current calibration process. Please check your wiring and try it again.

DwaineGarden, 2017-08-27:
I just was able to do a calibration on both channels.  I rebooted the psu and then did another calibration on each channel.  It was successful on both channels. During the current tests, the psu went into CC successful.  Ummmm.  Something not right and I guess it does not get reset unless a reboot.  

Len42, try a reboot and a Re-calibration

prasimix, 2017-08-27:
Yes, reboot is required after firmware upload if you talking about that case.

DwaineGarden, 2017-08-27:
Found the problem!!!  Looky what I have for a load.  It's not 1ohm, it's 1k.  The packaging was wrong!!!

Lol.    Len42, double check your load resistance. 

[image]

prasimix, 2017-08-27:
:rofl:

Len42, 2017-08-30:
You're right, I was not using the correct load. I didn't read the load requirements correctly.
Sorry about that. I will assume this is working correctly, until I get a proper load set up to do the calibration properly.

closed 2017-08-30 by Len42