eez-open/psu-firmware #161: Selftest failed on serial remote session (via Native USB) if error condition exists on channels

prasimix, 2017-08-28:
If error selftest is initiated via serial remote session (e.g. `SYST:POW 1` or `*TST?`) and one or more errors are detected on any channels a firmware will stuck and only hard reset can help. 
The easiest way to reproduce it is to simply remove one Power board, power up PSU, initiate serial session (via Native USB port!) and execute `*TST?`

mvladic, 2017-08-29:
Apparently, there is an limit of max. 64 characters when printing strings using `Serial.println(const char *)`. 

Following string was printed when CH2 temperature sensor test failed:

```
**ERROR [2017-08-29 13:03:14]: 723,"CH2 temperature sensor test failed"
```

which has 71 characters. That was cause of blocking.

In this fix, I replaced single `println` with multiple `print`'s followed by the `println` at the end.

prasimix, 2017-10-02:
Now it works but for some reason temp error is reported twice:

``` 
*TST?
**ERROR [2017-10-02 10:41:20]: 723,"CH2 temperature sensor test failed"
**ERROR [2017-10-02 10:41:20]: 723,"CH2 temperature sensor test failed"
**TRACE [2017-10-02 10:41:20]: Ch2 IO expander reg check failure: reg=0, expected=101, got=0
**ERROR [2017-10-02 10:41:20]: 211,"CH2 IOEXP test failed"
**TRACE [2017-10-02 10:41:20]: Ch2 ADC test failed reg1: expected=160, got=0
**TRACE [2017-10-02 10:41:20]: Ch2 ADC test failed reg2: expected=96, got=0
**ERROR [2017-10-02 10:41:20]: 221,"CH2 ADC test failed"
**TRACE [2017-10-02 10:41:20]: Ch2 DAC test skipped because of IO expander
**TRACE [2017-10-02 10:41:20]: Fan RPM: 3654
1
```

prasimix, 2017-10-02:
Fixed

closed 2017-10-02 by prasimix