eez-open/psu-firmware #162: Can't exit screen lock after reboot, date/time dialog present

pchickey, 2017-08-28:
* Firmware version: 1.0
* Shield version: Crowdsupply Backer
* Powerboard version: Crowdsupply Backer

### Expected behavior

Screen lock should be possible to exit by pressing on the lock icon.

### Actual behavior

Cannot exit screen lock by pushing lock icon, or rebooting. The "Do you want to set date and time?" dialog window is present on the screen, which may be related.

### How to reproduce

I assembled my Crowdsupply kit, went through the initial setup and calibration of both channels, and the device seemed to be working properly. I never set the date and time. I then entered screen lock, and then (a few minutes later) powered off the device with the standby switch on the front. When I powered it back on, the date dialog is on the center of the screen, but the screen is totally unresponsive - I can't do anything to get rid of the date/time dialog or unlock.

http://imgur.com/a/RPjaV

prasimix, 2017-08-29:
In the mean time you can use [SYST:LOC](http://www.envox.hr/eez/bench-power-supply/psu-scpi-reference-manual/psu-scpi-subsystem-command-reference/psu-scpi-system.html#syst_loc) command (using serial/USB connection) to unlock your display remotely.

pchickey, 2017-09-03:
Thanks. I just got around to trying SCPI tonight. I might be missing something obvious, but I'm unable to connect to the device over the usb/serial connection - with a USB cable connected to the Due "native" port, the Arduino shows up as a serial device, but when I send it any sort of input with a terminal (I'm using minicom, 9600 baud, 8n1) there is no response. Is there some recommended SCPI client besides using a dumb terminal? The docs don't seem to indicate otherwise.

prasimix, 2017-09-03:
Any text console should be fine for SCPI communications. Maybe you experience the same problem that is reported in [psu-hw #53](https://github.com/eez-open/psu-hw/issues/53). In that case you will have a problem with uploading firmware, too. 
Anyway, you can upload the latest firmware from master branch where this issue is solved.

pchickey, 2017-09-03:
I uploaded the firmware at 8fd8dfa and the issue is fixed. SCPI works now as well. Thank you for your help!

closed 2017-09-03 by pchickey