eez-open/psu-firmware #163: Both channel output is active during firmware updating. No indication that channels are active

DwaineGarden, 2017-08-30:
* Firmware version: latest git pull from master
* Shield version: R5B12
* Powerboard version: R5B12
[image]

The two output channels are active during a firmware update, even though the LCD and panel LEDs are off. 

The output channels should be off while doing a firmware update.  Attached is a photo of a firmware update that is uploading to the native port and the measured voltage on one channel.  I have tested both channels and they experience being active during the firmware update.

prasimix, 2017-08-30:
There is no cure for that since firmware uploading process is obviously not under firmware control :)

DwaineGarden, 2017-09-02:
Ok.... It still good to have this issue noted here for anyone which did not know that during the firmware update is the outputs are active, they'll be active during the firmware update process.

closed 2019-03-16 by prasimix