eez-open/psu-firmware #164: Unable to upload 1.01 firmware 'SerialUSB was not declared in this scope'

jcg747, 2017-09-09:
* Firmware version: 1.0
* Shield version : Arduino Due
* Powerboard : version  03/2017

I have current firmware 1.0 and tried to upgrade to 1.01
I have installed IDE 1.8.4 (windows version)
My EEZ H24005 is correctly connected on COM7 (native port)
When I press verify, I got the following error message
'SerialUSB was not declared in this scope'

Here is the full message
Arduino: 1.8.4 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
In file included from sketch\debug.cpp:19:0:
sketch\debug.cpp: In function 'void eez::psu::debug::DumpTraceBuffer()':
psu.h:53: error: 'SerialUSB' was not declared in this scope
 #define SERIAL_PORT SerialUSB
                     ^
sketch\debug.cpp:102:9: note: in expansion of macro 'SERIAL_PORT'
         SERIAL_PORT.print("**TRACE");
         ^
Multiple libraries were found for "Dhcp.h"
 Used: E:\Documents\Arduino\libraries\Ethernet2-1.0.4
 Not used: C:\Program Files (x86)\Arduino\libraries\Ethernet
 Not used: C:\Program Files (x86)\Arduino\libraries\Ethernet
 Not used: C:\Program Files (x86)\Arduino\libraries\Ethernet
 Not used: C:\Program Files (x86)\Arduino\libraries\Ethernet
exit status 1
'SerialUSB' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

I never played with arduino before, I'm just one who got this amazing powersupply through crowfunding.

I would appreciate any help for a newbie like me
Best regards from France
Jean Claude

prasimix, 2017-09-09:
Did you select _Arduino Due (Native USB Port)_ under **Tools... Boards:** in Arduino IDE? It seems that you have default board selected.

jcg747, 2017-09-09:
Thanks for the help Denis,
Argh, I followed the youtube firmware updater video so I selected Mega 2560 ...
So board is now : Arduino Due native port
port is COM9 (Arduino Due native port)
Get board info will bring:
BN: Arduino Due (Native USB Port)
VID: 2341
PID: 003E
SN: Upload any sketch to obtain it

When I press the verify button I get:
Arduino: 1.8.4 (Windows 10), Board: "Arduino Due (Native USB Port)"
Build options changed, rebuilding all
In file included from E:\Documents\Arduino\eez_psu_sketch\eez_psu_sketch.ino:23:0:
sketch\psu.h:1:28: fatal error: SoftwareSerial.h: No such file or directory
 #include <SoftwareSerial.h>
                            ^
compilation terminated.
exit status 1
Error compiling for board Arduino Due (Native USB Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

So what should I do ?

prasimix, 2017-09-09:
Are you trying to upload [1.01](https://github.com/eez-open/psu-firmware/releases/tag/1.01) or the latest firmware build from [master branch](https://github.com/eez-open/psu-firmware)?

jcg747, 2017-09-09:
Well , I just downloaded it from master branch. Should I download something else ?

jcg747, 2017-09-09:
I'm a noob,  I download the 1.01 and now I got this error message

Arduino: 1.8.4 (Windows 10), Board: "Arduino Due (Native USB Port)"
sketch\ethernet.cpp: In function 'void eez::psu::ethernet::init()':
ethernet.cpp:146: error: 'class EthernetClass' has no member named 'setDhcpTimeout'

     Ethernet.setDhcpTimeout(ETHERNET_DHCP_TIMEOUT * 1000UL);
              ^
exit status 1
'class EthernetClass' has no member named 'setDhcpTimeout'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

jlegen, 2017-09-09:
Did you follow '3.' from the installation instructions? Ethernet2 seems to be missing/wrong...

jcg747, 2017-09-09:
Sorry, where can I download installation instructions so I can see what '3' refer to ?

prasimix, 2017-09-09:
Everything is mentioned [here](http://www.envox.hr/eez/bench-power-supply/building-instructions.html#instr_fw) (that is section 3), maybe you didn't follow it from the start. So, _Ethernet2_ lib is available [here](https://github.com/eez-open/Ethernet2/releases/tag/1.0.2-eez).

jcg747, 2017-09-09:
Sorry for the mess, I will start from the beginning and will let you know.
Thanks again

jcg747, 2017-09-09:
Bingo, by downloading ethernet2 version 1.02-eez.zip instead of the 1.0.4 found on the web, everythin went fine and I have now firmware 1.01 loaded on this amazing product.

Again sorry for being so noob here but I never "touch" an Arduino before nor done any programming.
Thanks for you patience

closed 2017-09-09 by jcg747