eez-open/psu-firmware #165: Prepare .HEX for direct firmware upload without Arduino IDE

prasimix, 2017-09-16:
We should include .HEX file in firmware release as _texaspyro_ suggested [here](http://www.eevblog.com/forum/crowd-funded-projects/preparing-eez-h24005-for-crowdfunding/msg1284760/#msg1284760) that people who are not familiar with Arduino IDE can upload it to the PSU directly.

As code uploader for ARM (Arduino Due) we can probably use [BOSSA](https://github.com/shumatech/BOSSA/releases) that support main platforms (i.e. Win, Linux, OS X)

prasimix, 2017-09-16:
I tried to compile BOSSA 1.8 on Ubuntu 16.04LTS. It seems that the following two packages are needed:
`libwxgtk3.0-dev` and `libwxbase3.0-dev`. But even with them `make` returned the following error:

``` 
CPP BOSSA src/BossaForm.cpp
CPP BOSSA src/BossaWindow.cpp
CPP BOSSA src/BossaAbout.cpp
CPP BOSSA src/BossaApp.cpp
CPP BOSSA src/BossaBitmaps.cpp
CPP BOSSA src/BossaInfo.cpp
CPP BOSSA src/BossaThread.cpp
CPP BOSSA src/BossaProgress.cpp
LD bin/bossa
CPP BOSSASH src/bossash.cpp
src/bossash.cpp:33:31: fatal error: readline/readline.h: No such file or directory
compilation terminated.
Makefile:285: recipe for target 'obj/bossash.o' failed
make: *** [obj/bossash.o] Error 1
```

xorstream, 2017-10-18:
Perhaps a first step might to be to provide the prebuilt .HEX file in the releases/downloads area, either inside the source packs or as a separate file. Then once people have the .HEX file they can try programming it themselves and report back success or issues in doing so.

prasimix, 2018-04-29:
Will be added under release _Assets_ section

closed 2018-04-29 by prasimix

mrnuke, 2019-02-12:
I have bossa, a bricked eezpsu, the *.ino.elf, but still not sure how to put the three together.

prasimix, 2019-02-12:
If you are on Windows, it probably works fine, but first make sure that you have a "Arduino USB/serial" (or whatever is the name) driver installed. If I remember it correctly that driver is installed when Arduino IDE is installed. Other possibility is to follow this [instructions](https://www.arduino.cc/en/Guide/DriverInstallation). When driver is installed, the Arduino port should appear under "Serial port" list on the BOSSA. Select it, .ino under "File" and try Write. Possibly you should set some of "Write options". Please let me know if you succeed or not.

mrnuke, 2019-02-13:
I'm on linux. I tried

	$ ./bin/bossac -p /dev/ttyACM0 -e -w ~/Downloads/eez_psu_sketch.ino.elf 
	Erase flash
	Done in 0.023 seconds
	File operation exceeds flash size

Okay, so maybe bossa doesn't understand elf files.

	$ arm-none-eabi-objcopy -O binary eez_psu_sketch.ino.elf eez_psu_sketch.bin

	$ ./bin/bossac -p /dev/ttyACM0 -e -w eez_psu_sketch.bin 
	Erase flash
	Done in 0.023 seconds
	Write 439400 bytes to flash (1717 pages)
	[==============================] 100% (1717/1717 pages)
	Done in 12.591 seconds

You'd think that's a success, but it doesn't do anything. I know the board is not fried because it got bricked when connected to the retarduino IDE.

mrnuke, 2019-02-13:
Okay, so, the following procedure worked for me on linux:

	$ arm-none-eabi-objcopy -O binary eez_psu_sketch.ino.elf eez_psu_sketch.bin

	$ ./bossac -p /dev/ttyACM0 -e -b -w eez_psu_sketch.bin

Notice the **-b** parameter to bossac. Apparently that was the key.