eez-open/psu-firmware #171: Won't allow me to save calibration

reportingsjr, 2017-09-30:
* Firmware version: 1.01

### Expected behavior
When I get to the calibration remarks page, enter "new cal" and hit the checkmark button it should ask to save the calibration.

### Actual behavior
After hitting ok the calibration remarks page shows back up again and I don't get a prompt to save the calibration. Trying to hit the "next step" arrow says "invalid cal data" so my only other option is to exit calibration.

prasimix, 2017-09-30:
Is it possible for you to establish communication with serial and let us know how your trace log looks like during calibration? I presume that you passed all three phases successfully: voltage, high and low current.

reportingsjr, 2017-09-30:
All three phases were successful.

Here is what the serial log looks like starting from the front panel and ending at the "calibration remark" screen (using the Arduino 1.8.1 IDE with 115,200 baud rate):
**TRACE [2000-01-25 15:45:27]: Voltage range: 0.000000 - 40.000000V
**TRACE [2000-01-25 15:45:27]: ADC=0.121315
**TRACE [2000-01-25 15:46:41]: Current range: 0.000000 - 5.000000A
**TRACE [2000-01-25 15:46:41]: ADC=0.035878
**TRACE [2000-01-25 15:47:33]: ADC=0.003588
**TRACE [2000-01-25 15:47:44]: 

Also, I updated the firmware to master via the arduino ide over the native port. The issue stuck around.

prasimix, 2017-09-30:
Ok, why latest trace line is partial? Is that a copy&paste mistake or it's really like it? If I understood you correctly you cannot enter calibration text in the final step?

reportingsjr, 2017-09-30:
That is what it is really like. I thought it was strange as well.

I can enter the text, but when I hit the check mark to save the text it just goes back to the "calibration remark" screen and doesn't give me the "save calibration" modal. The calibration remark is shown there.

reportingsjr, 2017-09-30:
I was able to make it lock up at this point as well. After I entered the text and hit the checkmark, I then hit the home button, then hit "no", then when I tried to press the "next step" button it locked up until I restarted the PSU.

I'm not familiar with the DUE/Atmel SAM microcontrollers. Is there any way to run a debugger like gdb on these micros to step through code, set breakpoints, etc?

prasimix, 2017-09-30:
Strange. What's about other functions that require storing of parameters, e.g. saving user profile? Can you try to save current parameters under any user profile except 0? User profiles can be found under second icon on the bottom of the main page. See Fig 2 [here](http://www.envox.hr/eez/bench-power-supply/psu-firmware.html).

reportingsjr, 2017-09-30:
Oh, this is weird.

I connected the output of channel 1 to the current measurement terminal on my voltmeter (just so there would be a load) and when through the calibration. Rather than reading on the meter what the cal was I just entered exactly what it was trying to calibrate the PSU to and with those values it worked fine.

prasimix, 2017-09-30:
OK, so you still have a problem or not?

reportingsjr, 2017-09-30:
Tried the real calibration again and it wouldn't let me save cal again.

This is how I'm doing calibration fyi:

First the output is connected to the voltage measurement terminal and common terminal my bench voltmeter and I record the three calibration voltages.
Next I move the positive output from the voltage terminal to the current terminal and record the first two voltages.
Next for the high current range (just 5A) I move the positive output to my handheld meter since my bench meter can only do 3A.
Next for the low current range I move the positive output back to the current terminal on my bench meter.

I assumed this would be fine since even though the load is changing for the current measurements the PSU shouldn't care. The voltage during the current tests is thus pretty low (tens of mV up to about 2V). This should all be fine, right?

prasimix, 2017-09-30:
Ok, what is the value of your power resistor used for current calibration? It should be below 5 Ohms for high current range.

reportingsjr, 2017-09-30:
I was just using the current range (so basically their shunts) on my meters as the loads. My handheld meter is reading 4.97A during the test so I figured this was fine. I do have some power resistors below 5 Ohms I can use, just didn't want to measure the voltage across them to do the math for current. :P

A couple of other things to note (more information is better than no enough, right?):
I noticed during the voltage calibration that the current the PSU measures is negative.
This unit is the one you just shipped to me (bought via eevblog) so it has the new 10mOhm current shunt.

prasimix, 2017-09-30:
I'm not sure what you are doing. Please use for voltage calibration voltmeter and for current calibration external power load with ammeter (e.g. 10 A) connected in series with load. There is no need to calculate anything while calibrating: just enter values that you read on your voltmeter and ammeter.  
[Here](https://www.youtube.com/watch?time_continue=4&v=Wbq7mmp_3rc) you can also find a video about calibration (it doesn't cover latest revision with two current ranges, but still could be useful).

reportingsjr, 2017-09-30:
Ok, I followed that video exactly (besides for the lower current range) using a 3.8 ohm power resistor and it still did the same thing where it won't let me save calibration.

reportingsjr, 2017-10-01:
After quite a lot of debug statements (oh my god why does arduino not have a freaking debugger?) I narrowed it down to the low current range mid value check (Value::checkMid()). My unit was out of the allowed calibration tolerance (1% which seems a bit tight!). I changed the allowed tolerance to 10% which allowed my unit to pass. What I don't know is why the DebugTraceF in that function is failing and only printing out the time.

The values I entered for calibration were 4.4mA, 250mA, and 505mA if you wanted to try some testing to replicate this problem.

Let me know if there is anything I can do to help figure this out.

prasimix, 2017-10-01:
Thanks for that, Martin has to jump in here and instruct you how to proceed.

mvladic, 2017-10-02:
The reason why DebugTraceF didn't print out the whole data is for the same reason explained in this ticket:

https://github.com/eez-open/psu-firmware/issues/161

This is now fixed.

mvladic, 2017-10-02:
> I'm not familiar with the DUE/Atmel SAM microcontrollers. Is there any way to run a debugger like gdb on these micros to step through code, set breakpoints, etc?

I'm using [Atmer SAM-ICE](http://www.atmel.com/tools/atmelsam-ice.aspx) with [Ozone](https://www.segger.com/products/development-tools/ozone-j-link-debugger/).

prasimix, 2017-10-02:
> My unit was out of the allowed calibration tolerance (1% which seems a bit tight!). 

Such tight tolerance is set intentionally, if you'd like to have properly calibrated unit then 1% is more then enough with reasonably accurate external DMM.

closed 2017-10-05 by prasimix