eez-open/psu-firmware #176: Step mode value storage

AndersWestman, 2017-10-06:
Then changing values in step mode the size or “gain” of the steps follows between different settings.
E.g. I choose to change the voltage in 0.5V steps (next lowest step size) and then choose to change the current in 250mA steps (next highest). If I then goes back to set the voltage again, the step size now is 2V instead of 0.5V as it keeps the setting to the next highest value.
This may be a little surprising, especially then the rotary encoder is used.
Maybe the step size can be remembered for each value, or always be set to the lowest value then entering a new screen?

prasimix, 2018-04-06:
Works now per unit type (voltage, current) but not per channel. If we want to implement that correctly we have to remember 4 not just 2 steps.

prasimix, 2018-04-06:
Please try it now and close it if fixed.

AndersWestman, 2018-04-07:
Looks good.

closed 2018-04-07 by AndersWestman