eez-open/psu-firmware #180: Day saving change failed

prasimix, 2017-10-29:
It seems that day saving change failed tonight. My date/time settings are: NTP disabled, Time zone: +1 GMT, DST europe. Time was changed but difference is 5 hours, e.g. instead of 8:15 i have 3:15.

Note also that when NTP is disabled, time is not displayed in real time on settings page (I have to exit and enter page to get new time).

prasimix, 2017-10-29:
In the mean time I got another two `[02:00:00] Date/Time changed (DST)` message in event view. Time is again reset to 02:00:00 and counting! Probably we can expect new one every hour.

prasimix, 2017-10-30:
Still doesn't work or maybe I'm doing something wrong. Steps to reproduce:

1. Set date to 2017-10-29, and time to e.g. 02:59:30
1. Wait until 03:00 passed to check new time that is set to 02:00
1. Set time again to 02:59:30
1. Wait until 03:00 and time will be set once again to 02:00

Also it's questionable what time has to be recorded: old (03:00) or new (02:00)?

Refreshing of time on date and time settings page is still missing.

mvladic, 2017-10-30:
> 3. Set time again to 02:59:30

When you do this, DST is set again.

> Refreshing of time on date and time settings page is still missing.

Ok, we could refresh time but only until you make a change in any part of date and time, after which date and time will stay still.

closed 2017-10-30 by mvladic

prasimix, 2017-10-30:
> When you do this, DST is set again.

Does it mean that I have to wait one hour to see if everything works correctly?

prasimix, 2017-10-30:
> Ok, we could refresh time but only until you make a change in any part of date and time, after which date and time will stay still.

Sounds good!

mvladic, 2017-10-31:
> Does it mean that I have to wait one hour to see if everything works correctly?

Yes.

prasimix, 2017-10-31:
Tested, works as expected.

mvladic, 2017-10-31:
> Refreshing of time on date and time settings page is still missing.

Fixed!