eez-open/psu-firmware #183: Power limit warning can be overridden in serial coupling mode

AndersWestman, 2017-11-05:
* Firmware version: 1.01
* Shield version
* Powerboard version:  r5B12

### Expected behavior
Power limit warning are supposed to pretend settings over 310W.

### Actual behavior
In serial coupling mode power limit warning can be overridden if the PSU is in cc-mode and the load is small, below about 10 ohm, or shorted output.

### How to reproduce
Coupling: Serial
Volt: 80V
Load: 0-10 Ohm (shorted output works fine to reproduce this)

Enable output and increase the current in cc-mode.
At 3.876A power limit warning will occur, as expected. 
In step-mode or by rotary encoder it is however possible to go all the way up to 5A (400W) if it’s done with about max one step per second.

prasimix, 2017-11-05:
Additionally, when load is disconnected it is not possibly to change output values due to Power limit exceeded error.

AndersWestman, 2017-11-05:
More observersions, setup as described above.

Load 0-4.7 Ohm.
Keypad or SCPI-command can be used to go back under power limit level with no load or output off.

Current can be set from about 1.5-3A to anything up to 5A using keypad or SCPI-command without power limit warning. 
If current is set to something below 1-1.5A power limit warning occur as expected. This limit does not seem to be fixed. Sometimes I was able to jump directly from 1.1A to 5A without any warning, but sometimes I have to go from something like 2-3A. From 3A to 5A seems to always work.

With a 10 ohm load the PSU shut down due to “CH2 fault detected” then jumping from 3A to 5A using keypad.

prasimix, 2017-11-06:
Fixed

closed 2017-11-06 by prasimix