eez-open/psu-firmware #188: Fan speed control logic

kolesar007, 2017-11-28:
I have sugestion for fan speed logic:
we can have three settings: PWMmin and two temperatures Temp1 and Temp2. 

Speed PWMmin would be up to the Temp1.
Then between Temp1 and Temp2 PWM will be controled from PWMmin to 100%
From Temp2 fan will be at 100%

mvladic, 2018-03-27:
Does it mean that fan will be always on regardless of the temperature? If the main reason why you opened this issue is continuous fan switching on and off, then we has to find a way how to avoid it (e.g. by implementing PID controller).

kolesar007, 2018-03-28:
Yes fan is always on regardless of the temperature, and it would be nice to implement PID controler for fan. Now its anoying fan noise if you use power supply lets say just for small power.
As I wrote in my previous post I sugest set of two temperatures (temp1, tem2) in menu. So when temperature is below temp1 fan will be at min speed, between temp1 and temp2 fan will regulate from min speed to 100% and above temp2 fan will be at 100% of speed.

prasimix, 2018-03-28:
We'll add PID controller to manage fan speed. That should also offer possibility that fan works continuously (with various speed) depending of defined P, I, and D constants.

kolesar007, 2018-03-28:
You can add PID or simple as I wrote two temperatures and then drive PWM depend on this two temperatures.

prasimix, 2018-03-28:
Not sure that is the same: I don't want that fan is running all the time, and I believe it's possible to configure PID in both way (i.e. that fan is working continuously with different speed, or to switch off when measure temperature goes below one that is set for min. fan speed).

kolesar007, 2018-03-28:
Of course is not the same, but for that purpose I think that is fine simple control speed with temperature.
You have another option to regulate fan speed with current .. more current more speed :), but its better control with temperature because when you stop using supply with high current fan goes off but supply is still may be hot.

prasimix, 2018-04-05:
PID controller is now implemented and ready for testing. For debuging purpose you can change PID values without recompiling/uploading new firmware  using the SCPI command `DEBUG:FAN:PID Kp, Ki, Kd, poe` where:
* Kp = proportional
* Ki = integral
* Kd = differential and
* poe = process method, 1=_Proportional on Error_ or 0=_Proportional on Measurement_ (for more info check [Introducing Proportional On Measurement](http://brettbeauregard.com/blog/2017/06/introducing-proportional-on-measurement/)).

Current settings is `0.5, 0.05, 0, 1` that is infact PI controller since Kd=0. To check current settings use SCPI command  `DEBUG:FAN:PID?`

kolesar007, 2018-04-07:
Nice to hear that this is implemented :). I will test this next week.

closed 2018-04-29 by prasimix