eez-open/psu-firmware #189: Unable to compile sketch in Arduino: 'pages' not declared

richiewebgate, 2017-12-04:
I am trying to compile the psu-firmware-1.02 release in Arduino 1.8.5 but I am getting errors when verifying.

### Arduino output:

Arduino: 1.8.5 Hourly Build 2017/08/28 06:33 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
sketch\gui.cpp: In function 'void eez::psu::gui::init()':
gui.cpp:1175: error: 'pages' was not declared in this scope
     arduino_util::prog_read_buffer(pages, (uint8_t *)&g_documentBuffer, sizeof(Document));
exit status 1
'pages' was not declared in this scope

### What I have done so far: 
I followed the instructions in the psu-firmware readme:
- Downloaded eez_psu_sketch and moved libraries into respective folder
- Downloaded and installed additional libraries: Ethernet2 and SdFat
- Started Arduino and changed board to "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
- Unchecked "Verify code after upload" in file > preferences
- Changed CONF_SERIAL_USE_NATIVE_USB_PORT to 0 in conf_advanced.h to get rid of "SerialUSB" error

### Expected behavior

It should compile without errors when I click "Verify"

### Actual behavior

Throws error: gui.cpp:1175: error: 'pages' was not declared in this scope

### How to reproduce

Well, I did the same on a Ubuntu workstation and was also unable to compile. 

My PSU still has the original firmware on it, so it would be nice to be able to upgrade it to the most recent version. Maybe something is missing in the instructions?

mvladic, 2017-12-04:
> Started Arduino and changed board to "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Change board to Arduino Due (Native USB port). Before that you will need to download Arduino SAM Boards (check [this video](https://www.youtube.com/watch?v=Ar5kHiepsNM)).

richiewebgate, 2017-12-04:
Changing the board to "Due" did the trick! Also explains why I ran into the SerialUSB issue. Thanks a lot!

closed 2017-12-04 by richiewebgate