eez-open/psu-firmware #190: Ch1 Calibration current id

romain145, 2017-12-12:
* Firmware version:
* Shield version
* Powerboard version

### Expected behavior

### Actual behavior

### How to reproduce

prasimix, 2017-12-12:
Possible duplicate of #160. Trace log on serial is needed to check values.

romain145, 2017-12-13:
Current [0-5A] MID point:
 `**TRACE [2017-12-09 18:23:10]: Voltage range: 0.000000 - 40.000000V
**TRACE [2017-12-09 18:23:10]: ADC=0.191949
**TRACE [2017-12-09 18:24:13]: ADC check failed: level=2.425000, adc=-0.005169, diff=2.430169, allowedDiff=0.475000`

Current [0-5A] MAX point:
`**TRACE [2017-12-09 18:25:50]: ADC check failed: level=4.800000, adc=-0.005322, diff=4.805322, allowedDiff=0.475000`

Current [0-500mA] MID point:
`**TRACE [2017-12-09 18:27:03]: ADC check failed: level=0.242500, adc=-0.000502, diff=0.243002, allowedDiff=0.095000`

Current [0-500mA] MAX point:
`**TRACE [2017-12-09 18:27:55]: ADC check failed: level=0.480000, adc=-0.000486, diff=0.480486, allowedDiff=0.095000`

Channel 2 Current MID point:
`**TRACE [2017-12-09 18:29:17]: Voltage range: 0.000000 - 40.000000V
**TRACE [2017-12-09 18:29:17]: ADC=0.234834
**TRACE [2017-12-09 18:30:30]: ADC check failed: level=2.425000, adc=-0.005322, diff=2.430321, allowedDiff=0.475000`

Also, clicking "Change" after setting a calibration value doesn't re-enable the output.

prasimix, 2017-12-13:
Do you have load connected during current calibration?

romain145, 2017-12-13:
Yes, but it seems that my load impedance was way too low for the PSU.
I'm trying with a 1 Ohm load now and went through the calibration!

For some calibration, clicking "Change" after setting a calibration value doesn't re-enable the output.

prasimix, 2017-12-13:
> Yes, but it seems that my load impedance was way too low for the PSU.
> I'm trying with a 1 Ohm load now and went through the calibration!

You mean your load is too high, since measure current is zero. Use resistor that is lower then about 4 Ohms as it's clearly stated [here](http://www.envox.hr/eez/bench-power-supply/group-buy-kit-assembly.html#powering_up).

> For some calibration, clicking "Change" after setting a calibration value doesn't re-enable the output.

Please be more specific. Which "Change" button/option?

romain145, 2017-12-13:
The multimeter was correctly measuring the current so I'm not sure what the deal was...
Indeed, by using a 1-4 Ohm load, the calibration procedure runs just fine. Great job !

The Change option is working, I noticed blew the low amp range fuse on my multimeter (EEVBlog BM235). The multimeter case is rated 600mA, so I used it for the 500mA calibration; however, Dave fitted a 400mA fuse...

Closing

closed 2017-12-13 by romain145