eez-open/psu-firmware #197: Fan fail message during startup self-test when using a larger, slower fan.

morris6, 2018-03-16:
* Firmware version: v1.02 and v1.1
* Shield version: r5B13
* Powerboard version: r5B13

### Expected behavior

Self-test should be passed

### Actual behavior

Two out of three times self-test gives "Fan fail"

### How to reproduce

I used an eighty-millimeter fan in my build in stead of the prescribed sixty-millimeter one. It accelerates slower and has a lower nominal rpm, 2000 verses 4000.

### My investigations to find a solution

In the Fan.cpp file, I found the relevant functions and parameters; please correct me if I am wrong here.

The fan-test part of the self-test, function fan::test(), measures switches the fan on full speed and measures rpm after 250 ms. The measurement time is 25 ms (line 153 of original fan.cpp). In this period the speed signal must make a full cycle to be valid every time. That means at least 1200 fan rpm.

Furthermore, in function fan::tick(), the measurement time is 50 ms for a full cycle. Here fan speed lower than 600 rpm causes a fault condition during fan operating condition (line 241).

### My suggestion

Since there already exits a parameter FAN_NOMINAL_RPM, make use of this to control the measurement times. Introduce a new static variable dependent upon FAN_NOMINAL_RPM. Value could be 100000 / FAN_NOMINAL_RPM. Just take care that FAN_NOMINAL_RPM is not zero. Replacing the number 25 in test() with this and 50 in tick() with 2*this will clear the issue, FWIW.

### Tested

I did an adaptation of original fan.cpp file and 're' defined FAN_NOMINAL_RPM to 2000 in conf_user.h. This works for my build.

Included a zip containing copies of my modified files, for reference.
[Fan.zip](https://github.com/eez-open/psu-firmware/files/1819464/Fan.zip)

prasimix, 2018-04-05:
Ready for testing, please let us know if it works for you. Please also take into account changes introduced with #188.

morris6, 2018-04-05:
The adaptation of fan.cpp solves my issue. The startup self-test now performs as expected. Thanks.

closed 2018-04-05 by morris6