eez-open/psu-firmware #200: Using SCPI over usb serial then over ethernet causes a problem

ZakDanger, 2018-04-06:
* Firmware version: v1.1 (latest from github)
* Shield version: r5b12
* Powerboard version: r5b12

I am writing a little app to control the PSU via SCPI over usb serial and ethernet.
I have both usb serial and ethernet enabled on my PSU.
I have a usb cable and ethernet cable connected between my PSU and PC.
I have nothing connected to either channel of the outputs.

As a test I am just reading out the model via "SYST:CPU:MOD?\n".
When testing I will open the connection, send the command, read the result, and then close the connection. If I just use either serial OR ethernet, it works fine and I get the expected response of "Arduino, R5B12"\r\n.

If I perform this test over usb serial and then perform the test again over ethernet then it causes a problem. The usb serial test works fine, but then the following ethernet test does not return any result. If I then connect to the PSU via usb serial using a serial terminal app such as hyperterm it will immediately put my PSU into standby mode with the LCD screen off and BEEPs sounding on the PSU. Over the usb serial connection it will print:

**TRACE [2018-04-06 13:45:59]: A new ethernet client detected!
**TRACE [2018-04-06 13:51:21]: Ch1 PWRGOOD bit changed to 0
**ERROR [2018-04-06 13:51:21]: -242,"CH1 fault detected"
**TRACE [2018-04-06 13:51:21]: Ch2 PWRGOOD bit changed to 0
**ERROR [2018-04-06 13:51:21]: -243,"CH2 fault detected"
**TRACE [2018-04-06 13:51:21]: Profile 0 saved!
**TRACE [2018-04-06 13:51:21]: PWR_DIRECT -> LOW
**TRACE [2018-04-06 13:51:21]: Ch2 PWRGOOD bit changed to 0
**ERROR [2018-04-06 13:51:21]: -243,"CH2 fault detected"
**TRACE [2018-04-06 13:51:21]: Profile 0 saved!
**TRACE [2018-04-06 13:51:21]: PWR_DIRECT -> LOW
**TRACE [2018-04-06 13:51:21]: Ch1 PWRGOOD bit changed to 0
**ERROR [2018-04-06 13:51:21]: -242,"CH1 fault detected"
**TRACE [2018-04-06 13:51:22]: Ethernet client lost!

If I turn the PSU off and on again it will startup in standby mode and will print over usb serial:
EEZ PSU serial com ready
**TRACE: Touch screen calibration points: 3571, 445, 424, 3521, 421, 436
**TRACE: Touch modification ENABLED
**TRACE: Ethernet initialization started...
**TRACE: Listening on port 5025
My IP: 192.168.0.206
Netmask: 255.255.255.0
GW IP: 192.168.0.1
DNS IP: 192.168.0.1

If I then press and hold the LCD screen for a couple of seconds it will exit standby mode with both outputs turned off and print over usb serial:
**TRACE [2018-04-06 13:55:35]: PWR_DIRECT -> HIGH
**TRACE [2018-04-06 13:55:35]: Fan RPM: 4428
**TRACE [2018-04-06 13:55:37]: Profile 0 saved!

The PSU then continues to work normally.

I would expect it to be able to alternate between usb serial and ethernet connections like I am doing, especially because I am closing each connection before trying the next connection type.

To replicate:
1) Enable both usb serial and ethernet on your PSU.
2) Connect via usb serial using a terminal app on your PC and type "SYST:CPU:MOD?" (without quotes, and then press enter)
3) You should get a reply like "Arduino, R5B12"
4) Close the terminal app
5) Connect via ethernet using a telnet app on your PC and type "SYST:CPU:MOD?" (without quotes, and then press enter)
6) It won't give any reply
7) Close the telnet app
8) Connect via usb serial using a terminal app on your PC
9) Straight away it will print error messages to your terminal and put your PSU into standby mode.

mvladic, 2018-04-06:
Which terminal app are you using?

ZakDanger, 2018-04-06:
While testing repro steps I used SecureCRT in windows, but any terminal app should work for testing.
Initially the problem was occurring while using my own code, I just used a terrminal to provide repro steps and to ensure the issue wasn't due to my code.

After playing with it longer I think it may be connection open/close related inside the PSU firmware code as I get various random crashes when trying to talk to it over SCPI.

ZakDanger, 2018-04-06:
And by *crashes* I mean the PSU puts itself into standby mode as I mention in my initial post.

ZakDanger, 2018-04-06:
Unrelated to this issue: but I think your commit has an issue:
sd_card.cpp: line 67
is:
```g_fsBeginErrorCode = SD.fsBeginErrorCode();```
it looks like you probably meant:
```g_fsBeginErrorCode = SD.cardErrorCode();```

ZakDanger, 2018-04-06:
Ok so with the new code I can confirm that the serial terminal app I am using is not causing the usb icon to "turn off" when I disconnect from it. I see it go green when I connect to it, but it then stays green even when i disconnect.

ZakDanger, 2018-04-06:
Ah ok - i'll get it too then.
I just tried the code from my SCPI app, and the usb icon also stays green after a disconnect.

What do you do in your serial app to make the icon go off?
I am using pyserial, and when I am done I am calling serial.close() which is *supposed* to close the connection, but the green usb icon seems to say otherwise.

mvladic, 2018-04-06:
I'm using this library:

https://github.com/node-serialport/node-serialport

ZakDanger, 2018-04-06:
I just tried with the hyperterm app you linked (from google drive) it also does not turn off the usb icon.
The icon stays green.

mvladic, 2018-04-06:
I just tried pyserial on Windows and Ubuntu. On Windows it does not turn off usb icon but on Ubuntu it turns off!?

ZakDanger, 2018-04-06:
I am running windows 10 64bit if it matters.

Searching about closing serial port for arduino brings up lots of ppl also complaining about it, but I've not come across a fix yet.

ZakDanger, 2018-04-06:
I wonder if it is low level driver related - ie due to it being a usb connnection pretending to be a serial connection

mvladic, 2018-04-06:
I tried again hyperterm and it again turned off icon, strange why it doesn't work for you.

ZakDanger, 2018-04-06:
I am at 115200, 8N1.
I tried all 3 flow controls. no luck.

ZakDanger, 2018-04-06:
On ubuntu on my laptop I can get the icon to go off when I disconnect from it using putty.
Still yet to get it to go off while on my desktop running windows tho :/

ZakDanger, 2018-04-06:
Holy crap I just got it working!

After reading maaaany pages I some an advanced option for some usb drivers that lets you set RTS on close. So in my pyserial code I explicitly set RTS low before closing the connection normally and the green usb icon goes away!

Note I am not using dtr/rts flow control as part of my normal code.

For anyone else with this issue, here is my close code.
```
	def disconnect(self):
		if self.dev != None:
			self.dev.flush()
			self.dev.setRTS(0)
			self.dev.close()
			self.dev = None
```

mvladic, 2018-04-06:
I just find this article:

https://devzone.nordicsemi.com/b/blog/posts/weird-issue-with-pyserial-flow-control

it will also work if you open port like this:

```
ser = serial.Serial()
ser.port = 'COM1' # For example
ser.baudrate = 9600
ser.rtscts = True
ser.open()
```

ZakDanger, 2018-04-06:
I tried using all forms of flow control and none of them helped. In fact when I tried setting rtscts it would often result in the arduino not even responding, and when it did respond it would still not close the connection at the end. I tried every weird little hidden option as well along the way, nothing worked until I tried setting rts low just before closing. That appears to work every time.

ZakDanger, 2018-04-07:
I have tested again this morning, alternating between serial connection and ethernet connections. Now that the serial connection is closing correctly it doesn't crash anymore. So I will close this for now and reopen if the crashes come back :)

Thanks for the help tracking down the cause of this @mvladic

closed 2018-04-07 by ZakDanger