eez-open/psu-firmware #210: Rotary encoder issue

AndersWestman, 2018-08-23:
* Firmware version: 1.1.1 with the “SD-card fix”
* Shield version Crowd campaign unit
* Powerboard version

### Expected behavior
Rotary encoder is expected to change the value by one increment at time.

### Actual behavior
Rotary encoder sometimes changes the value by factor 100, e.g. 10V instead of 0.1V or 1V instead of 10mV.

### How to reproduce

prasimix, 2018-08-24:
How to reproduce: wait somewhere between half and hour and an hour after powering PSU on. First step will be ignored and the next one will be 100x times higher then expected (e.g. 1 V instead od 10 mV). All other steps after that will be correct (i.e. 10 mV increments).

AndersWestman, 2018-09-01:
Master branch from August 27, 2018 looks fine.

closed 2018-09-01 by AndersWestman