eez-open/psu-firmware #217: Cannot compile firmware for older boards

dsaltas, 2019-12-14:
* Firmware version: v1.1.2
* Shield version: R3B4
* Powerboard version: R5B9

### Expected behavior

I redefined conf_user.h to meet my boards.

/ /Redefine channels
#undef CHANNELS
#define CHANNELS \
    CHANNEL(1, CH_BOARD_REVISION_R5B9_PARAMS, CH_PINS_1, CH_PARAMS_40V_5A_R5B9), \
    CHANNEL(2, CH_BOARD_REVISION_R5B9_PARAMS, CH_PINS_2, CH_PARAMS_40V_5A_R5B9) \

// Select Arduino shield r3B4
#undef EEZ_PSU_SELECTED_REVISION
#define EEZ_PSU_SELECTED_REVISION EEZ_PSU_REVISION_R3B4

// Disable encoder
#undef OPTION_ENCODER
#define OPTION_ENCODER 0

// Disable SD card
#undef OPTION_SD_CARD
#define OPTION_SD_CARD 0
### Actual behavior

Compilation fails.

mvladic, 2019-12-14:
> Compilation fails.

Where exactly? Could you copy and paste compilation dump from Arduino IDE.

dsaltas, 2019-12-14:
This is the dump from Arduino IDE.
(The sketch compiles normally if I don't redefine user_conf.h) 

sketch/gui_data.cpp: In function 'eez::psu::gui::data::Value eez::psu::gui::data::get(const eez::psu::gui::data::Cursor&, uint8_t)':
gui_data.cpp:940:25: error: 'dlog' has not been declared
     if (!isInitiated && dlog::isInitiated()) {
                         ^
gui_data.cpp:950:22: error: 'dlog' has not been declared
     if (!isManual && dlog::g_triggerSource == trigger::SOURCE_MANUAL) {
                      ^
gui_data.cpp:1241:31: error: too few arguments to function 'bool eez::psu::list::anyCounterVisible(uint32_t)'
   if (list::anyCounterVisible()) {
                               ^
In file included from sketch/gui_data.cpp:29:0:
sketch/list.h:71:6: note: declared here
 bool anyCounterVisible(uint32_t totalThreshold);
      ^
Multiple libraries were found for "SPI.h"
 Used: /Users/jimmy/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/libraries/SPI
Multiple libraries were found for "scpi-parser.h"
 Used: /Users/jimmy/Documents/Arduino/libraries/scpi-parser
Multiple libraries were found for "SdFat.h"
 Used: /Users/jimmy/Documents/Arduino/libraries/SdFat
Multiple libraries were found for "eez_psu_rev.h"
 Used: /Users/jimmy/Documents/Arduino/libraries/eez_psu_lib
Multiple libraries were found for "Ethernet2.h"
 Used: /Users/jimmy/Documents/Arduino/libraries/Ethernet2
exit status 1
'dlog' has not been declared

mvladic, 2019-12-14:
Indeed there was a couple of problems when compiling R3B4. It should be fixed now, I just pushed the fix. Please get the latest commit.

Also, I can see some syntax errors in your conf_user.h. I don't know if they are introduced only during copy and paste, but to be sure here is the correct version:

```
// Redefine channels
#undef CHANNELS
#define CHANNELS\
CHANNEL(1, CH_BOARD_REVISION_R5B9_PARAMS, CH_PINS_1, CH_PARAMS_40V_5A_R5B9), \
CHANNEL(2, CH_BOARD_REVISION_R5B9_PARAMS, CH_PINS_2, CH_PARAMS_40V_5A_R5B9) \

// Select Arduino shield r3B4
#undef EEZ_PSU_SELECTED_REVISION
#define EEZ_PSU_SELECTED_REVISION EEZ_PSU_REVISION_R3B4

// Disable encoder
#undef OPTION_ENCODER
#define OPTION_ENCODER 0

// Disable SD card
#undef OPTION_SD_CARD
#define OPTION_SD_CARD 0
```

dsaltas, 2019-12-14:
It compiles now, the syntax errors introduced during copy & paste.
Many thanks!

closed 2019-12-14 by dsaltas